mcci-catena/arduino-lmic #4: LMIC code is inconsistent on max frame length

terrillmoore, 2017-04-02:
* `lorabase.h` defines `MAX_LEN_FRAME` as 64, and this value is used most places (either as such or as MAX_PAYLOAD_LEN).
* `lmic.h` defines `MAX_FRAME_LEN` as 64, and this is never used (but it has a doxygen tag as an API value).
* `radio.c` calls `writeReg(LORARegPayloadMaxLength, 64);` to set the max frame length.
* `lmic.c` has a macro `maxFrameLen()` that returns the max frame size for a given data rate. However, that macro is not used.
* `lmic.c` checks the size of data, before copying into LMIC.pendTxData[], but then doesn't check against the data rate being used.

There seems to be no good reason that a larger frame can't be supported; the LMIC code uses the fifo in half-duplex mode, so with a larger frame buffer, bigger messages could be handled (up to the 250 byte max defined in the spec). But there also doesn't seem to be a good reason that the length is not validated against the current uplink rate before calling LMIC_setTxData().

christopherdro, 2018-06-13:
Related https://github.com/mcci-catena/arduino-lmic/issues/77

closed 2019-03-19 by terrillmoore

mrx23dot, 2022-02-23:
Ideally 
- user should specify MAX_USER_PAYLOAD len in config 
- lib should range check it
- add extra hdr len
- finally write it into LORARegPayloadMaxLength 

this would also eliminate wasting RAM.

commit fb50600d (2019-03-11): Fix #4: make MAX_FRAME_LEN an alias for MAX_LEN_FRAME