mcci-catena/arduino-lmic #8: Minor debugging output changes and printf() support

jcwren, 2017-04-21:
I always manage to screw up git pull requests, so I hope you don't mind a diff file (I'm a Subversion guy).

This is against commit  ab06411.

Adds printf() support on non-AVR platforms (works on Feather M0)
Adds missing timestamps on three debug prints.

```
diff -r arduino-lmic-master/src/hal/hal.cpp arduino-lmic-jcw/src/hal/hal.cpp
264a265,266
> #if defined(__AVR)
> // avr-libc provides an alternative (simpler) way to override STDOUT
281a284,307
> #else // defined(__AVR)
> //
> // On other platforms, use the somewhat more complex "cookie"-based
> // approach to custom streams. This is a GNU-specific extension to libc.
> //
> static ssize_t uart_putchar (void *uart, const char *buf, size_t len)
> {
>     return ((Uart *) uart)->write (buf, len);
> }
>
> static cookie_io_functions_t functions =
> {
>     .read = NULL,
>     .write = uart_putchar,
>     .seek = NULL,
>     .close = NULL
> };
>
> void hal_printf_init()
> {
>     if ((stdout = fopencookie (&LMIC_PRINTF_TO, "w", functions)))
>       setvbuf (stdout, NULL, _IONBF, 0);
> }
> #endif // !defined(__AVR)
304a331,334
> #if 1
>     printf("FAILURE %s:%u\n", file, (int)line);
>     fflush(stdout);
> #else
309a340
> #endif
diff -r arduino-lmic-master/src/lmic/lmic.h arduino-lmic-jcw/src/lmic/lmic.h
33a34
> #include <stdio.h>
diff -r arduino-lmic-master/src/lmic/radio.c arduino-lmic-jcw/src/lmic/radio.c
616c616
<   LMIC_DEBUG_PRINTF("start single rx: now-rxtime: %lu\n", now - LMIC.rxtime);
---
>   LMIC_DEBUG_PRINTF("%lu: start single rx: now-rxtime: %lu\n", now, now - LMIC.rxtime);
624c624
<         LMIC_DEBUG_PRINTF("RXMODE_RSSI\n");
---
>         LMIC_DEBUG_PRINTF("%lu: RXMODE_RSSI\n", os_getTime ());
841c841
<       LMIC_DEBUG_PRINTF("rxtimeout: entry: %lu rxtime: %lu entry-rxtime: %lu now-entry: %lu rxtime-txend: %lu\n", entry, LMIC.rxtime, entry - LMIC.rxtime, now2 - entry, LMIC.rxtime-LMIC.txend);
---
>       LMIC_DEBUG_PRINTF("%lu: rxtimeout: entry: %lu rxtime: %lu entry-rxtime: %lu now-entry: %lu rxtime-txend: %lu\n", os_getTime (), entry, LMIC.rxtime, entry - LMIC.rxtime, now2 - entry, LMIC.rxtime-LMIC.txend);
```

terrillmoore, 2017-04-23:
@jcwren : Thanks for the patches.

Can work with a diff file, but it will take me a while to get to it (partially because it's a manual process, and partially because it's the end of the month).

Best regards,
--Terry

terrillmoore, 2017-09-18:
Now that we have a single branch again (with multi-region support), I will try to look at this again. However, it still may be a few weeks before this bubbles to the top.

commit 04b5789f (2018-10-14): Merge pull request #151 from mcci-catena/issue150

Fix #150: add CI testing for debug prints. Also fix #8, making debug prints work on gcc-based non-AVR.
closed 2018-10-14 by terrillmoore