mcci-catena/arduino-lmic #21: LMIC_setDrTxpow() is ineffective

terrillmoore, 2017-07-14:
`LMIC_setDrTxpow()` doesn't really do anything.

Although calling it sets `LMIC.adrTxpow` to the input parameter, the value is then never used inside the LMIC library. This is because LMIC's radio.c uses `LMIC.txpow`, and in US915, `lmic.c::updatetx()` sets `LMIC.txpow` to 30 for 125kHz channels, and  26 for 500kHz channels, ignoring `LMIC.adrTxpow`.  Finally, radio.c limits to the value for 10 dBm, and apparently doesn't even turn on the +20 dBm option if over 10 dBm.

So it seems that neither manual nor automatic TX power control can set the power above 10 dBm; and (in the US), the TX power is fixed at 10 dBm. 

* fix the handling of > 10 dBm in radio.c
* fix the  US `updatetx()` so that ADR works
* test ADR in US, EU, etc.

terrillmoore, 2018-02-07:
See #50 #46 etc.

ngraziano, 2018-11-27:
For EU like region we can do something like this commit ngraziano@3948c8c94b61a6b5ca10e215a07e3539a0d805ed

It will allow the network to reduce power if quality of signal is good enough (TTN do that).

ngraziano, 2018-11-28:
I check regional parameter and I see a problem with region like AS923 where Max EIRP is set by `TxParamSetupReq`.
In actual code for AS923, `adrTxPow` store the delta and not the absolute value.

Maybe `adrTxPow` should store the delta for all region and not the absolute value.

But I am little lost because in lorawan regional parameters for E868 do not match the code and there is no power per band.

This topic need more thinking.

pomplesiegel, 2019-01-26:
Any thoughts on how this would be resolved? Any quick hard-coded fixes for those of us using this only one region and trying to LIMIT power? In my case, I would like to put the RFM95W to its lowest power level, but calls like
```
LMIC_setDrTxpow(DR_SF7,2);
```
and 
 
```
LMIC.datarate = DR_SF7;
LMIC.txpow = 2; 
```
seemingly still have no effect.

Thank you!

pomplesiegel, 2019-02-07:
Update: I'm just manually setting 
```cpp
s1_t pw = 2; 
```
in radio.c, as my region and location will be static. Still though, is there a plan to resolve 
`LMIC_setDrTxpow()` and restore its effectiveness? Am I correct in believing that it does not affect the Spread factor as well? 

Thank you!

terrillmoore, 2019-02-09:
@pomplesiegel -- `LMIC_setDrTxpow()` sets the spreading factor, but that will get reset by network downlinks that happen during and after a joint.  In other words, you may find that you have to set the data rate several times to make things effective, and you may have to turn off ADR (which can be done, and which works). Network operators are not always happy if you do this, but... sometimes it's necessary.

As for changing the tx power, yes... the delay is due to lack of time on my part to set up a proper test -- we really want to make sure we're controlling power properly.

I have been planning to run US LoRaWAN certification tests, which would (I think) clear up enough of the otehr issues to let this be addressed. But I don't know if I'll get to it this quarter, because of pressing work for MCCI's other projects.

pomplesiegel, 2019-02-10:
@terrillmoore, thanks so much for the in-depth response. This is very helpful information on the spreading factor, as I had heard varying reports from others. 

I totally understand about the tx power issue and the need to get it right. In the meantime people like me will just hard-code a value and that's fine. 

Very exciting about the US LoraWan cert tests. I imagine that is a major question on many of our minds: When following best practices and using sanctioned hardware with this fork of LMIC, are we being compliant and future-proof with our devices? 

Thanks again!

terrillmoore, 2019-09-04:
This is now fixed with #416 and #410

closed 2019-09-04 by terrillmoore

Commifreak, 2020-10-22:
I have an additional question about the Tx power... What is the initial value if nothing is set? I use the lib for an SX1276 with this config:

```c
#define CFG_eu868
#define CFG_sx1276_radio 1
#define DISABLE_PING
#define DISABLE_BEACONS

#define LMIC_DEBUG_LEVEL 2
```

Thanks :)

martin3000, 2020-11-17:
see lorabase_eu868.h :
EU868_TX_EIRP_MAX_DBM = 16