mcci-catena/arduino-lmic #26: LMIC doens't implmement RX1DRoffset

terrillmoore, 2017-08-07:
It doesn't capture it from the JoinAccept, and it always acts as if it were zero.

terrillmoore, 2017-09-18:
As of 29b0eea8, the code captures `LMIC.rx1DrOffset` from the JoinAccept message. As of b87bab1e5b31bd0832435a7a527f5f1ac2a8e686, the as923 region correctly incorporates `LMIC.rx1DrOffset` into the rx1 window calculation -- this is in important for as923 because of the uplink/downlink dwell bits. However, have not reviewed other regions still.

GuyWittig, 2018-02-08:
The RXParamSetupReq code in lmic.c does not process the RX1DRoffset bits either.

terrillmoore, 2019-03-14:
As of 29b0eea8a2e, this was fixed for AU921. As of 833e525, this is fixed for US915.

terrillmoore, 2019-03-14:
As of 2be3c52, this is fixed for EU868 and IN866.

commit 833e525c (2019-03-14): Fix #26 for US915: calculate rate
commit 2be3c526 (2019-03-14): Fix #26 for eu868 and in866
closed 2019-03-19 by terrillmoore