mcci-catena/arduino-lmic #33: How to lock the frequency when having a single channel gateway?

jrnker, 2017-09-26:
Hi,
I saw an issue in the original repo (https://github.com/matthijskooijman/arduino-lmic/issues/120) which I am wondering about as well.

I am, just as that guy, using a single channel gateway, and am in need of locking my node to a fixed channel. As discussed in that thread, I've been trying to lock out the unwanted channels with `LMIC_disableChannel(x)` without success, and resorted to a nasty quickfix to get it up running.

How do I achieve this, to reduce the wanted channels to a single one, preferably to the one where the join negotiation successfully happened?

Should `LMIC_disableChannel(x)` be used in a specific way to achieve this?

Cheers,
Chris

terrillmoore, 2017-09-26:
Hi,

 

Which region are you using (EU, US, etc.)?

 

Best regards,

--Terry

terrillmoore, 2017-09-26:
Are you doing OTAA or ABP?

 

For ABP, after initializing the LMIC you should call `LMIC_disableChannel()`.

 

For OTAA, you need to call `LMIC_disableChannel()` immediately after starting a JOIN, and then again after a successful JOIN.  (Starting a JOIN resets the mask.)

 

The JOIN-ACCEPT message sets the channel mask according to what the network wants, and that doesn’t cater to single-channel gateways.  So you have to reset the mask again.

 

In addition you have to worry more about duty cycle limits, etc.

 

Best regards,

--Terry

RickardPettersson, 2017-09-27:
@terrillmoore  i have tried the code below but not getting it to work more then the joing works over 868.1 and then next package sents on other channel. 

Any idea if i missing to do the channel stuff somewhere?

`
#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

// This EUI must be in little-endian format, so least-significant-byte
// first. When copying an EUI from ttnctl output, this means to reverse
// the bytes. For TTN issued EUIs the last bytes should be 0xD5, 0xB3,
// 0x70.
static const u1_t PROGMEM APPEUI[8]= {  };
void os_getArtEui (u1_t* buf) { memcpy_P(buf, APPEUI, 8);}

// This should also be in little endian format, see above.
static const u1_t PROGMEM DEVEUI[8]= {  }; 
void os_getDevEui (u1_t* buf) { memcpy_P(buf, DEVEUI, 8);}

// This key should be in big endian format (or, since it is not really a
// number but a block of memory, endianness does not really apply). In
// practice, a key taken from ttnctl can be copied as-is.
// The key shown here is the semtech default key.
static const u1_t PROGMEM APPKEY[16] = {  }; 
void os_getDevKey (u1_t* buf) {  memcpy_P(buf, APPKEY, 16);}

static uint8_t mydata[] = "Hello, world!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 60;

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 8,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 4,
    .dio = {7, 6, LMIC_UNUSED_PIN},
};

void onEvent (ev_t ev) {
    Serial.print(os_getTime());
    Serial.print(": ");
    switch(ev) {
        case EV_SCAN_TIMEOUT:
            Serial.println(F("EV_SCAN_TIMEOUT"));
            break;
        case EV_BEACON_FOUND:
            Serial.println(F("EV_BEACON_FOUND"));
            break;
        case EV_BEACON_MISSED:
            Serial.println(F("EV_BEACON_MISSED"));
            break;
        case EV_BEACON_TRACKED:
            Serial.println(F("EV_BEACON_TRACKED"));
            break;
        case EV_JOINING:
            Serial.println(F("EV_JOINING"));
            break;
        case EV_JOINED:
            Serial.println(F("EV_JOINED"));

            SetChannels();
            
            // Disable link check validation (automatically enabled
            // during join, but not supported by TTN at this time).
            LMIC_setLinkCheckMode(0);
            break;
        case EV_RFU1:
            Serial.println(F("EV_RFU1"));
            break;
        case EV_JOIN_FAILED:
            Serial.println(F("EV_JOIN_FAILED"));
            break;
        case EV_REJOIN_FAILED:
            Serial.println(F("EV_REJOIN_FAILED"));
            break;
            break;
        case EV_TXCOMPLETE:
            Serial.println(F("EV_TXCOMPLETE (includes waiting for RX windows)"));
            if (LMIC.txrxFlags & TXRX_ACK)
              Serial.println(F("Received ack"));
            if (LMIC.dataLen) {
              Serial.println(F("Received "));
              Serial.println(LMIC.dataLen);
              Serial.println(F(" bytes of payload"));
            }
            // Schedule next transmission
            os_setTimedCallback(&sendjob, os_getTime()+sec2osticks(TX_INTERVAL), do_send);
            break;
        case EV_LOST_TSYNC:
            Serial.println(F("EV_LOST_TSYNC"));
            break;
        case EV_RESET:
            Serial.println(F("EV_RESET"));
            break;
        case EV_RXCOMPLETE:
            // data received in ping slot
            Serial.println(F("EV_RXCOMPLETE"));
            break;
        case EV_LINK_DEAD:
            Serial.println(F("EV_LINK_DEAD"));
            break;
        case EV_LINK_ALIVE:
            Serial.println(F("EV_LINK_ALIVE"));
            break;
         default:
            Serial.println(F("Unknown event"));
            break;
    }
}

void SetChannels() {
  LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  1);      // g-band
  LMIC_setupChannel(1, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  1);      // g-band
  LMIC_setupChannel(2, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  1);      // g-band
  LMIC_disableChannel(3);
  LMIC_disableChannel(4);
  LMIC_disableChannel(5);
  LMIC_disableChannel(6);
  LMIC_disableChannel(7);
  LMIC_disableChannel(8);
}

void do_send(osjob_t* j){
    // Check if there is not a current TX/RX job running
    if (LMIC.opmode & OP_TXRXPEND) {
        Serial.println(F("OP_TXRXPEND, not sending"));
    } else {
        // Prepare upstream data transmission at the next possible time.
        LMIC_setTxData2(1, mydata, sizeof(mydata)-1, 0);
        Serial.println(F("Packet queued"));
    }
    // Next TX is scheduled after TX_COMPLETE event.
}

void setup() {
    delay(2000);
    Serial.begin(9600);
    Serial.println(F("Starting"));

    #ifdef VCC_ENABLE
    // For Pinoccio Scout boards
    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
    delay(1000);
    #endif

    // LMIC init
    os_init();
    
    // Reset the MAC state. Session and pending data transfers will be discarded.
    LMIC_reset();
    
    SetChannels();
    
    
    // Start job (sending automatically starts OTAA too)
    do_send(&sendjob);
}

void loop() {
    os_runloop_once();
}`

terrillmoore, 2017-09-28:
Looks to me like you’re enabling three channels in SetupChannels(). Please check. Here’s the code in question:

```c++

void SetChannels() {
    LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7), 1); // g-band

    LMIC_setupChannel(1, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7), 1); // g-band
    LMIC_setupChannel(2, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7), 1); // g-band
    LMIC_disableChannel(3);
    LMIC_disableChannel(4);
    LMIC_disableChannel(5);
    LMIC_disableChannel(6);
    LMIC_disableChannel(7);
    LMIC_disableChannel(8);
}

```

Don’t change the default setting (at least at first). Instead just allow the defaults to propagate. Try this.

```c++

void SetChannels() {

   // leave setting of channel alone, and turn off others.

   for (unsigned ch = 1; ch <= 15; ++ch) {

      LMIC_disableChannel(ch);

   }

}

```

RogerDeng, 2017-11-09:
Hi,
I have the same question in AS923 region. I already define CFG_as923 in code. but output frequency isn't within 923-925Mhz range. (ex: 907.7,902.3,908.9 ....) , why?

In TTN LoRaWAN Frequencies information (https://www.thethingsnetwork.org/wiki/LoRaWAN/Frequencies/Frequency-Plans). AS923-925 uplink total 10 channel. If I choose only use 923.2Mhz to send data .After I use LMIC_disableChannel(ch) to disable 1 to 10 channels but doesn't work. How should I do? 

Thanks

terrillmoore, 2017-11-09:
The theory for AS923 is that until Join, only 3 channels are used. After Join, the JoinResponse, the network sends the channel frequencies. What gateway and core network are you using? We have tested with the Things Network configured for AS923 and a MultiTech gateway. In this configuration, it worked well for us.

However, we have not tested ABP. Are you using OTAA or ABP?  Actually, ABP requires that you manually set up all the channels (other than the fixed initial channels). It's possible that there's a bug that we just don't see. Can you give more info on your test scenario?

RogerDeng, 2017-11-10:
Yes, I use ABP.  

I use DRAGINO LG01-P 915Mhz version (http://www.dragino.com/products/lora/item/117-lg01-p.html) to set up my 923.2Mhz 1ch LoRaWAN test environment and I use this version (https://github.com/orenonline/arduino-lmic-as923) limc library with AS923 modification. I can LMIC_disableChannel to left only 923.3Mhz. It's work.  But I use your limc-1.6 version. LMIC.freq always show incorrect frequency. Hope can give you some useful information.

avbentem, 2019-01-05:
See also https://www.thethingsnetwork.org/forum/t/can-lmic-1-6-be-set-up-to-use-a-single-channel-and-sf/5207/11

terrillmoore, 2019-09-04:
The current version forbids changing the default channels. As all regions have multiple default channels, single channel gateways will not be compatible unless you do some clever hacking with datacrate masks and linkadrreqs. Goal of this project is compliance.

closed 2019-09-04 by terrillmoore