mcci-catena/arduino-lmic #36: SX1278 (433Mhz) configuration

vshymanskyy, 2017-12-05:
Is there any way I can configure it to work with SX1278 + single-channel gateway (my gateway seems to be working already) ?
I tried this, but it didn't help:
```
LMIC_setupChannel(0, 434500000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);
```
Thanks!

terrillmoore, 2017-12-05:
Hi,

Wow, I really don't know; I've not tried this at all. What I've found is that the code is pretty fragile in the face of overrides like that, but if you set up a proper region it seems to work OK. Which LoRaWAN region are you trying to use?

Best regards,
--Terry

terrillmoore, 2017-12-05:
By the way: there is documentation (untested by others, but used by me during the conversion) on adding bandplans: [arduino-lmic/HOWTO-ADD-REGION.md](https://github.com/mcci-catena/arduino-lmic/blob/master/HOWTO-ADD-REGION.md). Even if you don't want to do the work of adding a bandplan, reading through it might give a clue as to what could be going wrong.

--Terry

vshymanskyy, 2017-12-05:
Thanks. I'll try to figure it out!

closed 2017-12-05 by vshymanskyy