mcci-catena/arduino-lmic #37: SPI.begin() variants

vshymanskyy, 2017-12-06:
On some boards, SPI.begin() is used to set SPI pins:
```
SPI.begin(5,19,27,18);
```
Currently, this is hardcoded in the library, but it may be best to leave it for the user sketch.
https://www.thethingsnetwork.org/forum/t/big-esp32-sx127x-topic/10247/131

cyberman54, 2018-10-21:
@terrillmoore I could need this, too, for my paxcounter software which supports generic boards.

cyberman54, 2018-10-21:
Suggested implemenation with pull request #157

terrillmoore, 2018-10-28:
Unfortunately, I discovered that my suggestion of adding to the pinmap can't work on Adafruit BSPs. Sigh. Plus (after suggesting that) I learned that changing the pinmap can be a major breaking change. See my comments on #157.

terrillmoore, 2019-02-25:
This is closed by #190.

closed 2019-02-25 by terrillmoore