mcci-catena/arduino-lmic #40: Starting os_init (); problem

manibenwa, 2017-12-11:
Hai ,
Iam new to lora,
Iam using this example https://github.com/mcci-catena/arduino-lmic/tree/master/examples/raw-feather and  i configure config.h file to (#define CFG_in866 1 and #define CFG_sx1276_radio 1) and iam using the adafruit M0 board (https://www.adafruit.com/product/3178).
My compilation success ,after uploading the code ,i got  the following error
Starting
FAILURE 
/home/neelgreev/Arduino/libraries/arduino-lmic-master/src/lmic/oslmic.c:55
can u please tell me where is the problem??where should modification occurs in code??

terrillmoore, 2017-12-12:
Usually this means that the pin configuration table is wrong. The error message comes from an ASSERT(), and the ASSERT is happening because `radio_init()` returned false. The line numbers don't match exactly, so I don't think you're using exactly the code from the repo, or perhaps not the most recent version; but it looks pretty conclusive anyway.

You should not be modifying `config.h`.  See https://github.com/mcci-catena/arduino-lmic#configuration, where it says:

>A number of features can be enabled or disabled at compile time. This is done by adding the desired settings to the file `project_settings/lmic_project_config.h`. The `project_settings` directory is the only directory that contains files that you should edit to match your project; we organize things this way so that your local changes are more clearly separated from the distribution files. The Arduino environment doesn't give us a better way to do this.
>
> Unlike other ports of the LMIC code, in this port, you should not edit `src/lmic/config.h` to configure this package.

By the way, did you add the required jumper on the Feather M0 LoRa? It's not as clear in README.md as it should be, but the docs say:

> This pin table assumes that you have manually wired JP1 pin 1 to Arduino JP3 pin 9 (Arduino D6).

This is required. (But the problem it fixes doesn't arise until you far past initialization; your problem is arising during initialization.)

If you're still having problems, perhaps our Chennai team (@svelmurugan92 and group) can help.

Best regards,
--Terry

manibenwa, 2017-12-12:
Thanks for replay,
I have not wired any thing (JP1 pin 1 to Arduino JP3 pin 9 (Arduino D6))
can u please tell me which pins should i need to connect in lora feather M0??

terrillmoore, 2017-12-12:
Please wire Feather M0 LoRa **JP1 pin 1** to Feather M0 **JP3 pin 9**.  Refer to the [schematic](https://cdn-learn.adafruit.com/assets/assets/000/032/914/original/feather_schem.png?1465421956) for background info.

Here's a sketch (on the back side -- not the component side -- of the board). Note that if you are using the MCCI Catena 4450/4460 Wings, the boards already provide this wire so you don't need to do this. In all other cases, this wire is needed. [image]

manibenwa, 2017-12-13:
I have connected as you show in above figure,still i stuck on the same error,
by the way iam using this lora (https://www.thingbits.net/products/adafruit-feather-m0-rfm69hcw-packet-radio-868-or-915-mhz)  and it has SX1231 module  and i followed 

const lmic_pinmap lmic_pins = {
    .nss = 8,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 4,
    .dio = {3, 6, LMIC_UNUSED_PIN},
};
but still facing same error.can you send me the most resent version of this library(https://github.com/mcci-catena/arduino-lmic/tree/master/examples/raw-feather )

terrillmoore, 2017-12-13:
Oh, that’s the problem. The SX1231 module doesn’t work with Arduino LMIC. You must have the SX1272 or SX1276. Sorry!

closed 2017-12-15 by terrillmoore

yogeshmadi, 2017-12-19:
Thanks a lot for your useful information@terrillmoore. Can you please tell me is there any Lorawan library for SX1231 or any other similar library which can support this SX1231???

Can you tell me why this board doesn't support Lorawan protocol(LMIC Library)??  or is any hardware or software issue???

terrillmoore, 2017-12-19:
It's different hardware. The modulation for LoRa is unique, and can't be handled with a normal FSK radio. The SX1231 is not a LoRa-capable radio.

Best regards,
--Terry

yogeshmadi, 2017-12-19:
Thanks for the reply@terrillmoore. Here iam using this https://www.adafruit.com/product/3176 board and this board have 868 or 915mhz radio frequency so please suggest me any related libraries is there for this board???

yogeshmadi, 2017-12-21:
Hai   terrillmoore,

actually we have this coaster pagers https://www.aliexpress.com/item/Fast-food-Wireless-Restaurant-Guest-paging-system-KTP-20-coaster-pager/32597068448.html 

Can u please tell which protocol used in this coaster system???