mcci-catena/arduino-lmic #56: 'const lmic_pinmap' has no non-static data member named 'rxtx_rx_polarity'

DavidBoelee, 2018-03-24:
Hi,

This may me a noob question, but when I try to compile an LMIC example sketch with the suggested settings for the Catena 4551 board, I am getting the following error:

`const lmic_pinmap' has no non-static data member named 'rxtx_rx_polarity`

I am using this settings:

`
const lmic_pinmap lmic_pins = {
    .nss = 7,
    .rxtx = 29,
    .rst = 8,
    .dio = {25, 26, 27},
    // the Murata module needs D29 high for RX, low for TX.
    .rxtx_rx_polarity = 1, 
    // the Murata module is direct-wired, we can use 8 MHz for SPI.
    .spi_freq = 8000000
};`

When I comment out the `    .rxtx_rx_polarity = 1, ` the code is compiled without errors.

Is this a bug, or am I overlooking something? I did install the Catena boards extension in my IDE.

DavidBoelee, 2018-03-24:
Update: I searched back and installed exactly the specified version: 2.1.0. The error is still there and same for later versions.

terrillmoore, 2018-03-24:
Sorry you're having problems. Let me take a look; I don't usually use the examples, and we don't have CI set up, so it's possible that we missed updating this during the revision that completed the support for hte Murata module.

terrillmoore, 2018-03-24:
Hi again,

I researched this. Very sorry, this is a documentation error.

The file `libraries/arduino-lmic/src/hal/hal.h` defines a field named named "rxtx_rx_active", not "rxtx_rx_polarity".  It has the documented function.

I strongly suggest you update to 2.1.4, because there are some important minor changes. If you're using our board support package, also update to v1.0.8 (if using Linux or macOS -- v1.07 will work well for Windows).

Sorry for the inconvenience. I'll update the docs now.

--Terry

closed 2018-03-24 by terrillmoore

terrillmoore, 2018-03-24:
Thanks very much for your report. Docs are updated and new release posted.

DavidBoelee, 2018-03-25:
Thanks for getting into this and solving it so quick! Now it all makes sense. I already wondered why I couldn't find "rxtx_rx_polarity" in the source code at all.

commit 82a89f31 (2018-03-24): Fix #56: erroneous pinmap name in README.md
commit 9491bf9b (2018-03-24): Merge pull request #57 from mcci-catena/tmm-issue56

Fix #56: erroneous pinmap name in README.md