mcci-catena/arduino-lmic #60: 868 band BAND_MILLI, BAND_CENTI not defined

DavidBoelee, 2018-03-25:
When I use the LMIC library with the 868 band configured in `lmic_project_config`, I get an error when I compile the ABP example. The error indicates that  BAND_MILLI and BAND_CENTI are not configured. Am I supposed to configure them myself in my sketch, or is this an omission in the library?

After some searching, I had the impression that  BAND_MILLI should be 0, and BAND_CENTI should be 1. When I hard code these numbers in the example sketch, I can indeed send data in combination with ABP.

More detailed information about these constants would be greatly appriciated!

terrillmoore, 2018-03-25:
I am sorry to say that we have not tested ABP much at all with MCCI's version of the library. In my work with TTN NYC and TTN Ithaca, I found that ABP was really a headache, so focused on OTAA. But this error is just a compile error.

With respect to your problem, the scope of `BAND_MILLI` got changed, and it's not in scope anymore from `arduino_lmic/src/lmic.h`.  The simple fix in `src/lmic.h` (the name is confusing, but historical.

```c++
// after this line:
#include "lmic/lmic.h"
// add this line:
#include "lmic/lmic_bandplan.h"
```

Clearly I need to configure continuous integration. I'll file a bug for that, too.  I'll commit an patch to master but may hold off on a release for a day due to other pressing issues.

DavidBoelee, 2018-03-25:
Thanks! For now I can solve this myself, so no need for an early release.

terrillmoore, 2018-03-25:
code is changed on master. I'll do a release after allowing my group at MCCI to do some testing.

commit 44e2112a (2018-03-25): Fix #60: src/lmic.h needs lmic/lmic_bandplan.h
commit 7bbcb2c6 (2018-03-25): Merge pull request #62 from mcci-catena/issue60

Fix #60: src/lmic.h needs lmic/lmic_bandplan.h
closed 2018-03-25 by terrillmoore