mcci-catena/arduino-lmic #80: EV_TXSTART missing from Event Handler

trlafleur, 2018-06-29:
This is missing from your event handler in the ttn-otaa.ino example (maybe others)

Its now caught as an Unknown event

terrillmoore, 2018-06-29:
Thanks for reminding me about this. I'll update the sample apps.

closed 2018-06-30 by terrillmoore

trlafleur, 2018-06-30:
This might of interest you....  It swaps byte to allow pasting of keys
direct from TTN

// **********************************************************
// Function to do a byte swap in a byte array
void RevBytes(unsigned char* b, size_t c)
{
  u1_t i;
  for (i = 0; i < c / 2; i++)
  {
     unsigned char t = b[i];
    b[i] = b[c - 1 - i];
    b[c - 1 - i] = t;
  }
}

/*
**************************************************************************************
*/
// provide device ID (8 bytes, LSBF)
void os_getDevEui (u1_t* buf) {
    memcpy(buf, DEVEUI, 8);
    RevBytes(buf, 8);      // TTN requires it in LSB First order, so lets
swap byte's
}

/*
**************************************************************************************
*/
// provide application router ID (APPEUI) (8 bytes, LSBF)
void os_getArtEui (u1_t* buf) {
    memcpy(buf, APPEUI, 8);
    RevBytes(buf, 8);      // TTN requires it in LSB First order, so lets
swap byte's
}

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

commit 56ac1e5f (2018-06-30): Fix #80: add printout for EV_TXSTART (and clean up)
commit 9f07d483 (2018-06-30): Merge pull request #84 from mcci-catena/issue79

Fix #80: add EV_TXSTART in examples; clean up for CI