mcci-catena/arduino-lmic #88: useing debug

trlafleur, 2018-06-30:
Is there a version of the code that works with LMIC_DEBUG_LEVEL > 0, or insight on how to implement this?   ie: printing out debug info on M0 or ESP32 processor on Arduino??

I have tried define these without any luck...
#define LMIC_DEBUG_PRINTF_FN
#define LMIC_DEBUG_PRINTF

thanks...

terrillmoore, 2018-08-29:
Very sorry for the slow response on this. It got missed.

Obviously, more guidance is needed. I regularly use this on other (non-Arduino) platforms, but I have not done it recently on Arduino. I think that Frank Rose uses this in his projects - he's given some pull requests. Check https://github.com/frankleonrose especially https://github.com/frankleonrose/LoraStack for examples that might be useful.

--Terry

terrillmoore, 2018-10-14:
Fixed by #150. Sorry for the delay. We now have CI tests that will confirm that this doesn't regress.

closed 2018-10-14 by terrillmoore