mcci-catena/arduino-lmic #96: support for EU433 band ?

ccmsi, 2018-08-29:
Greetings
I see wide, predefined support for frequency bands, with the exception of 433MHz (EU433). I this available / not mentioned ? If not available any hint what /where in the code I should look at to make it work would be appreciated.
Thanks.

terrillmoore, 2018-08-29:
Hi @ccmsi, thanks for your note.

We haven't done it primarily because of test headaches -- we don't have any device hardware with 433 MHz support in house.  Since we work a lot with the Feather M0, we could use https://www.adafruit.com/product/3179 and do some testing. But that may take awhile.

I documented the procedure for adding a region: see [HOWTO-ADD-REGION](https://github.com/mcci-catena/arduino-lmic/blob/master/HOWTO-ADD-REGION.md). Feel free to comment on deficiencies or areas of needed improvement....

Once we can test, it will be great to add this to the distribution.

Best regards,
--Terry

saspol, 2019-07-16:
Hi all, i "added" eu433 region with this hint:
http://wiki.dragino.com/index.php?title=Connect_to_TTN#How_To_add_other_frequency_band_such_as_EU433
And looks like it working ok with my gateway.

andruby, 2020-06-02:
@saspol would you mind sharing the exact changes that made it work on 433Mhz?

I can help you create a pull request so we can merge it into the library for other users

saspol, 2020-06-02:
@andruby sorry, no patch, just dirty hack lorabase_eu868.h

```
enum {
        EU868_F1 = 868100000,      // g1   SF7-12
        EU868_F2 = 868300000,      // g1   SF7-12 FSK SF7/250
        EU868_F3 = 868500000,      // g1   SF7-12
        EU868_F4 = 868850000,      // g2   SF7-12
        EU868_F5 = 869050000,      // g2   SF7-12
        EU868_F6 = 869525000,      // g3   SF7-12
        EU868_J4 = 864100000,      // g2   SF7-12  used during join
        EU868_J5 = 864300000,      // g2   SF7-12   ditto
        EU868_J6 = 864500000,      // g2   SF7-12   ditto
};
enum {
        EU868_FREQ_MIN = 863000000,
        EU868_FREQ_MAX = 870000000
};
*/
enum { EU868_F1 = 433175000,      // g1   SF7-12
       EU868_F2 = 433375000,      // g1   SF7-12 FSK SF7/250
       EU868_F3 = 433575000,      // g1   SF7-12
       EU868_F4 = 433775000,      // g2   SF7-12
       EU868_F5 = 433975000,      // g2   SF7-12
       EU868_F6 = 434175000,      // g3   SF7-12
       EU868_J4 = 434375000,      // g2   SF7-12  used during join
       EU868_J5 = 434575000,      // g2   SF7-12   ditto
       EU868_J6 = 434775000,      // g2   SF7-12   ditto
 
};
enum { EU868_FREQ_MIN = 433050000,
       EU868_FREQ_MAX = 434900000
};
```

> @saspol would you mind sharing the exact changes that made it work on 433Mhz?
> 
> I can help you create a pull request so we can merge it into the library for other users

terrillmoore, 2020-06-02:
Not surprising that it sort of works, but there is more to do (which is why I've not done it).

I sketched the procedure for adding a region: [HOWTO-ADD-REGION.md](https://github.com/mcci-catena/arduino-lmic/blob/master/HOWTO-ADD-REGION.md).

Basically, you need to compare the 433 regional requirements (section 2.5 of the 1.0.3 regional spec) with the 868 regional requirements, and then do the work outlined in the procedure to create a new region. EIRP must be limited to 12.15dBM. There is a 1% duty-cycle restriction in 433 (much simpler than 868). There's no dwell-time limitation. The max payload sizes are the same. The TXPower field only goes 0..5 (rather than 0..7)  The beacon frequency is different.

By the way, EU868_J4, J5, J6 are not used - they should be removed from the EU868 plan. Ditto F4 and F5; but F6 is used as the beacon frequency. This is wrong for EU434, as they use a different frequency (434.665) as the beacon and downlink pingslot frequency.

Armed with this kind of info, it's straightforward to duplicate the EU868 and create EU434. The key thing is to get the 1% duty-cycle restriction by putting all channels into the CENTI group. I think the AS923 region might be a good example of this; there's no need for the MILLI stuff, or the fancy checking in `LMIC_setupChannel()` to find 0.1%, 1% and 10% duty-cycle groups, as apparently the 434MHz band allows up to 10% duty cycle.

gioreva, 2024-08-31:
Hi
I need EU433. 
I have the hardware to do tests with Avr CPU. 
Do you want to add it? 
I will do the tests.