mcci-catena/arduino-lmic #104: Include Hello-World for Sensors with 2.2.0

brentru, 2018-09-19:
Library should have an example to demonstrate sending a multi-value payload from a sensor using the `LMIC_f2sflt16` function added in 2.2.0.

terrillmoore, 2018-09-19:
Good idea. I'd do it based on one of MCCI boards, but that is probably assuming more infrastructure.  We could measure Vbat and send it, that's really very cool, and that will work on the Feather M0 stand-alone. If you have a lib for the CPU temperature, we could use that too. Check out the Vbat traces as part of https://ithaca-power.mcci.com/ -- you'll need to sign up, but it's permissive. It's very exciting to watch the battery discharge (oddly enough).

brentru, 2018-09-19:
I'm working on one for a learnguide I'm writing with a DHT22 and a Feather M0 LoRa. I'll do a PR when ready.

terrillmoore, 2018-09-19:
Fixed by #108.

closed 2018-09-19 by terrillmoore