mcci-catena/arduino-lmic #114: Error compiling library from include, missing "scalbnf"

lunarHonour, 2018-09-23:
I am getting an error, using master branch. 

...
C:\Users\x\Documents\Arduino\libraries\arduino-lmic-master\src\lmic/lmic_util.c:88: undefined reference to `scalbnf'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board LoRa32u4II (868-915MHz).

lunarHonour, 2018-09-23:
Tested master release 2.1.5, does not suffer from same error. I believe it has to do with the new code that converts data to sflt16. Not sure why my math.h does not have scalbnf defined. Using Arduino 1.8.7 IDE with a LoRa32u4II board.

terrillmoore, 2018-09-23:
Yes, that makes sense. I don't have a regression test case for the AVRs. Very sorry.  The quick workaround for you would be to delete the new lmic_util.c file as it's not used by the LMIC library but is provided for users. (I take it that the AVR compiler doesn't do as good a job of omitting unused code, or that <math.h> is incomplete.)

terrillmoore, 2018-09-23:
Hi again... I tested with the Adafruit 32u4 BSP without difficulty. So I can't repro this. I'm using 1.8.5 with Adafruit BSP; test ws ttn-otaa-feather-us915.ino (after editing to add `|| defined(ARDUINO_AVR_FEATHER32U4)` at line 77). This looks like a BSP limitation. Can you give me the exact link to the BSP you're using?

lunarHonour, 2018-09-24:
I think I am using an older version of this https://github.com/BSFrance/BSFrance-avr so my board is defined as "BSFrance AVR Boards" ->  "LoRa32u4II (868-915MHz)" 
Let me try and switch to https://github.com/adafruit/Adafruit_Arduino_Boards (feather32u4) and report back.

terrillmoore, 2018-09-24:
Oh. I see you're compiling `ttn-otaa-feather-us915-dht22.ino`. That was the key missing bit of information. I regret to say that probably `scalbnf()` is not available in the AVR libraries. It is possible that `ldexp()` is available, and will work. I can't test, other than doing a compile test, but I'll check that.

terrillmoore, 2018-09-24:
Yep. Please try this, since you can test (I can only compile):

1. open libraries/arduino-lmic/src/lmic/lmic_util.c
2. add (after `#include <math.h>`, line 18):

   ```c++
   #define scalbnf(x,e)	ldexpf(x,e)
   ```

3. Compile (works for me)
4. See if you're able to send data.

In case you're finding strange results, you should do some experiments in a sketch:

```c++
#include <math.h>
...
for (int i = 0, i < 16; ++i)
   Serial.println(ldexpf(1.0, i));
```

You should get a table of powers of 2 (1, 2, 4, 8, 16, etc., up to 65536).

lunarHonour, 2018-09-24:
It compiles/links now! Wohoo!

" for (int i=0; i<16; ++i)
    Serial.println(ldexpf(1.0, i));"

Results in:

1.00
2.00
4.00
8.00
16.00
32.00
64.00
128.00
256.00
512.00
1024.00
2048.00
4096.00
8192.00
16384.00
32768.00

For some reason I am getting a "0" result on thingsnetwork using that javascript decoder "function sflt162f(rawSflt16)".

terrillmoore, 2018-09-24:
Please try `for (int i=0; i < 16; ++i) Serial.println(ldexpf(1.0, -i));` -- make sure you get 1, 0.5, 0.25, etc.

Then look carefully at how you're feeding the decoder function. Remember that you have to pick up two bytes and combine them before calling the decoder. Brant transmitted the data low-order byte first, so you'll need to compute `buf[i] + buf[i+1]*256`, in this example, and pass that number to the decoder.

I'm going to merge my fix #128 to master now. No time for a release tonight.

lunarHonour, 2018-09-24:
NVM, 

I did not have full decoder code as I did not realize that it worked in two byte pairs... just found this code:

`function Decoder(bytes, port) {
  // Decode an uplink message from a buffer
  // (array) of bytes to an object of fields.
  var decoded = {};
  
  // temperature 

  rawTemp = bytes[0] + bytes[1] * 256;
  
  decoded.degreesC = sflt162f(rawTemp) * 100;
  
  // humidity 
  rawHumid = bytes[2] + bytes[3] * 256;
  decoded.humidity = sflt162f(rawHumid) * 100;
  
  return decoded;
}

function sflt162f(rawSflt16)
	{
	// rawSflt16 is the 2-byte number decoded from wherever;
	// it's in range 0..0xFFFF
	// bit 15 is the sign bit
	// bits 14..11 are the exponent
	// bits 10..0 are the the mantissa. Unlike IEEE format, 
	// 	the msb is transmitted; this means that numbers
	//	might not be normalized, but makes coding for
	//	underflow easier.
	// As with IEEE format, negative zero is possible, so
	// we special-case that in hopes that JavaScript will
	// also cooperate.
	//
	// The result is a number in the open interval (-1.0, 1.0);
	// 
	
	// throw away high bits for repeatability.
	rawSflt16 &= 0xFFFF;

	// special case minus zero:
	if (rawSflt16 == 0x8000)
		return -0.0;

	// extract the sign.
	var sSign = ((rawSflt16 & 0x8000) != 0) ? -1 : 1;
	
	// extract the exponent
	var exp1 = (rawSflt16 >> 11) & 0xF;

	// extract the "mantissa" (the fractional part)
	var mant1 = (rawSflt16 & 0x7FF) / 2048.0;

	// convert back to a floating point number. We hope 
	// that Math.pow(2, k) is handled efficiently by
	// the JS interpreter! If this is time critical code,
	// you can replace by a suitable shift and divide.
	var f_unscaled = sSign * mant1 * Math.pow(2, exp1 - 15);

	return f_unscaled;
	}
`
from  https://learn.adafruit.com/the-things-network-for-feather/arduino-code

and it works fully. 

Thank you!!

terrillmoore, 2018-09-24:
that's great. Will close this now. BTW: has Adafruit already published the Learning Center article? Or did you just stumble upon this?

closed 2018-09-24 by terrillmoore

lunarHonour, 2018-09-24:
Just stumbled on it from a google search of `function sflt162f(rawSflt16)`
 [https://www.google.com/search?q=function+sflt162f%28rawSflt16%29&ie=utf-8&oe=utf-8&client=firefox-b-1-ab](url)