mcci-catena/arduino-lmic #130: node hangs after "Packet queued" message

frazar, 2018-09-24:


terrillmoore, 2018-09-24:
I take it that this is not "The Things Network" as the network server? Definitely agree that this is a problem, but I need more context, as I've not seen this in my testing. (Need to figure out what additional test coverage is needed.)

terrillmoore, 2018-09-25:
OK, cool! Another test case.... Interesting to think about how one could do hardware-in-the-loop testing with this.

You will probably find that this version of the LMIC stack doesn't do a great job with MAC command responses. See issue #87 for details. TL;DR: it doesn't built the response as it's parsing the request, so things can go back out of order (or not at all).

PR #131 (2018-09-27): Fix coding of the "Margin" field of the "DevStatusAns" command payload
The value of 31 is taken from this [line](https://github.com/brocaar/lorawan/blob/f8d816eb916de4ef3253cc6f1f1036d25a1e324b/mac_commands.go#L563) of the lorawan software.

Fixes #130
commit d9225fb8 (2018-09-27): Merge pull request #131 from frazar/set_max_margin_to_31

Fix #130: Fix coding of the "Margin" field of the "DevStatusAns" MAC command payload
closed 2018-09-27 by terrillmoore