mcci-catena/arduino-lmic #132: LMIC_disableChannel(15) doesn't work?

joeynoodlesfang, 2018-09-26:
Hi,

I'm using the US915 bands, and using a RFM95+ESP32, coupled with a pycom LoPy single channel gateway. I'm running the abp file. I'm currently forced to use the 905.3MHz band. This is because when I try to disable channel 15, I receive:

Starting
RXMODE_RSSI
8817: engineUpdate, opmode=0x808
FAILURE
C:\.......\MCCI_LoRaWAN_LMIC_library\src\lmic\lmic_us_like.c:162

I have adapted the channel isolating code to:

```
int channel = 3;
int dr = DR_SF7;

void forceTxSingleChannelDr() {
    for(int i=0; i<15; i++) { // For EU; for US use i<71
        if(i != channel) {
            LMIC_disableChannel(i);
        }
    }
    for(int i=16; i<64; i++) { // For EU; for US use i<71
        if(i != channel) {
            LMIC_disableChannel(i);
        }
    }
    // Set data rate (SF) and transmit power for uplink
    LMIC_setDrTxpow(dr, 14);
}
```

Channel 3, in this case, was the channel I intended on using. However, that never happens, it's always channel 15 that gets used.

The problem is not urgent, as I will eventually move onto an actual gateway. But I was curious as to where the problem actually is.  Any suggestions are greatly appreciated.

terrillmoore, 2018-09-27:
When do you issue this? The assert says that *no* 125kHz channel is enabled. You don't enable channel 3. During a JOIN in the US,  TTN sends a channel mask (which disables all channels except 8 through 15, "subband 1" (zero origin). Since you don't call LMIC_enableChannel(channel), you're going to end up with all 125kHz channels disabled. The ASSERT() is on the bug-list; we should be returning errors, rather than asserting, but this is definitely not a valid situation.

terrillmoore, 2019-03-19:
I've heard nothing about this for a while; I am going to close this. Please reopen if it's still a problem.

closed 2019-03-19 by terrillmoore