mcci-catena/arduino-lmic #145: KR921 Frequency Plan

pctj101, 2018-10-09:
I've written up some code for a KR921 frequency plan.  Would you be interested in having me submit it to be part of this project?

I'd like to get your feedback on what kind of copyright notice should go into new files based on existing files and how much validation you need before merge.  I've tested and join/data up/downstream works but it's easy to miss some fine detail since I'm not a LoRa expert.

https://github.com/manuelbl/ttn-esp32/pull/7

Some projects require testing to expert levels before contribution, while others will accept "something that works for me".   I'm not able to do expert level testing, so I'm trying to figure out if I'd have enough time to make a proper contribution.

terrillmoore, 2018-10-14:
Hi again @pctj101 -- I reviewed your pull request https://github.com/manuelbl/ttn-esp32/pull/7, and it looks generally fine. I see a few typos from the original comments that came across when you duped the files, which we can fix in review if you submit a pull request, but this looks ready to go, code-wise. Feel free to add your name or organization to the list of copyrights on the new files.

pctj101, 2018-10-14:
Thanks for the check. I’ll send something over in the next day or two

terrillmoore, 2019-08-15:
HI @pctj101 -- I didn't see input from you, and I had an urgent MCCI request to add kr920. So .. I duplicated effort, alas, because I'd completely forgotten about this. But kr920 is now in place #360, with docs. I've run a compliance test with the RedwoodComm 5020A and things work about as well as they do in the other regions. Need to fix #87 before things mostly pass, but that's a common problem.

terrillmoore, 2019-09-04:
This is now complete and passes LoRaWAN precertification tests on the RWC 5020A

closed 2019-09-04 by terrillmoore

pctj101, 2019-09-04:
@terrillmoore Wow nice!  Thanks for the update!  I wish I could have paid more attention, but got a bit swamped in the meantime.