mcci-catena/arduino-lmic #185: Defaults flags in `lmic_project_config.h` cannot be overwritten in PlatformIO setup

frazar, 2018-11-23:
I'm trying to use the `arduino-lmic` library within a PlatformIO project. 

My `platformio.ini` looks like this:

```
[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
lib_deps =
    MCCI LoRaWAN LMIC library=https://github.com/mcci-catena/arduino-lmic/archive/master.zip
build_flags =
    -D CFG_eu868=1
    -D LMIC_DEBUG_LEVEL=2
    -D LMIC_PRINTF_TO=Serial
```

The `LMIC_DEBUG_LEVEL` and `LMIC_PRINTF_TO` flags work as expected.
However, the code fails to compile because the `CFG_eu868` flag clashes with the `CFG_us915` flag defined by `lmic_project_config.h`. 

The complete error message is the following:
```
In file included from .piolibdeps/MCCI LoRaWAN LMIC library/src/aes/ideetron/../../lmic/oslmic.h:36:0,
from .piolibdeps/MCCI LoRaWAN LMIC library/src/aes/ideetron/AES-128_V10.cpp:44:
.piolibdeps/MCCI LoRaWAN LMIC library/src/aes/ideetron/../../lmic/config.h:25:3: error: #error You can define at most one of CFG_... variables
# error You can define at most one of CFG_... variables
^
```

I see a number of solutions:
1. Leave everything as-is, and force the user to copy paste the library in the `src` dir of the project. However, this is tedious, error-prone and prevents the user from taking advantage of the extremely rich dependency management that is provided by PlatformIO.
2. Prevent `lmic_project_config.h` from defining the `CFG_us915` if other flag is already specified.
3. Remove all the flag definitions from `lmic_project_config.h`, and expect the user the desired flags.

terrillmoore, 2018-11-23:
Sorry, I don't support platformio -- I couldn't make it work at all for me. I know others like it, but... my focus in this platform is on the Arduino environment. I believe that @frankleonrose made PlatformIO work by doing all the -D's from the comand line. one will prevent the lmic_project_config.h from being included. But we can't change lmic_project_config.h at this time.

frazar, 2018-11-24:
While investigating the problem, I found that the library already supports a `ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS` flag which prevents `lmic_project_config.h` from being included!

closed 2018-11-24 by frazar

lnlp, 2019-06-27:
### lmic_project_config.h should contain comments only

Why?: 

1. lmic_projects_config.h is meant for PROJECT specific settings. 
_Your defaults_ are not applicable for many of your library's users, e.g. not everyone is living in the US.

2. For above reason all users will have to check and many will have to modify lmic_projects_config.h anyhow. 
It's is better to generate a compilation error that the user has not added certain configuration parameters than to assume (add) some default values which are incorrect for many users.
This is error prone.

3. This is the third but most most important reason: 
Adding (inappropriate) default #definitions prohibits and hinders the use of smart / more advanced automatic tooling, because it is very hard to impossible to automatically undefine because they are unknown because they were 'hard-coded' into the source of a library.
One of the things that it makes impossible are the build configuration features in PlatformIO's PROJECT CONFIGURATION FILE.
Project configuration files should be defined on project level with project scope, but not on a lower level in 'some' library's source code but latter is exactly what lmic_project_config.h does (this are unwanted dependencies).

I understand that the Arduino IDE has several major handicaps and that this sometimes requires sub-optimal choices. I also understand the reasoning behind lmic_project_config.h and that it may be a workable workaround for the Arduino IDE. However, having to modify the source code of a library for changing some configuration parameters is bad practice and should be prevented where possible because it introduces maintenance nightmares and prohibits the use of automated tooling.
And in case of lmic_project_config.h this can be prevented by having it contain no active statements and only comments (which can be uncommented).

Why is adding default definitions to lmic_project_config.h a problem for PlatformIO users?:

@frazar actually already hinted about this.

* PlatformIO uses a project specific PROJECT configuration file. It allows to define flexible settings which can automate most of the work for the user, including settings for the build proces which also includes defining the LoRaWAN region settings. These settings are done at project level and do not require any tweaking of library sourcecode (unless the library's design makes this impossible).
* In platformIO (again in the project configuration file) you can simple define the NAMES of libraries that are used (and a specific version if that is required) and PlatformIO will automatically download those libraries for the user, without any further user intervention. The libraries are downloaded into a project specific directory (.piolibdeps) that is normally not touched by the user.
If the user WANTS to make any changes to a library because he/she feels need to (which is different than being FORCED to make changes for configuration parameters) then the user puts the library MANUALLY in a different directory (lib). But then the library is not automatically updated anymore (another of PlatformIO's handy features).

Whyle @cyberman54 proposed a nice workaround for redefining the location of lmic_project_config.h this is awkward and unnecessary. It forces users to do a workaround that should not be necessary in the first place. 

So these above things are made impossible for PlatformIO users (users of your library) when MCCI's LMIC library adds any default settings to lmic_project_settings.h. And it can be very easily prevented by commenting them out.
This make the PlatformIO community happy while still also serving the Arduino IDE users and even makes them aware that they have to conciously select the region for their LoRaWAN application.

> "Sorry, I don't support platformio."

Major part of the Arduino community (the users of your library) is using PlatformIO these days because it makes things 10 times easier. Keep in mind that users of your library are the more advanced users who prefer the best development tools (otherwise they would not be developing LoRaWAN devices/application with LMIC, and else they would probably still use the original Arduino-LMIC library) and not your improved version.
Please have a serious look at PlatformIO again. It works and it works great.
I am sure that you will like and appreciate it (like MANY users of your library).
There will be enough people who can help you make a jumpstart.

** PLEASE comment out all active statements in lmic_project_config.h.**
Even if you do not support PlatformIO you should not implicitly provide anti-support and make things impossible.

Thanks

terrillmoore, 2019-06-27:
Thanks for your comments. Please study the flag -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS, which is intended for use with PlatformIO, and which causes all definitions to come from the command line.