mcci-catena/arduino-lmic #195: Spread Factor 10 default at AS923

3nd0y, 2018-11-26:
Hi Devs,

It's good to hear MCCI support adding AS923 which is will use in our country (Indonesia), here i have a bit question. Every time joining OTAA, is it absolutely the node will use SF 10 and will never use other SF? How to change default SF?

Here log attached..

> Packet queued
> 161031584: engineUpdate, opmode=0x8
> 161031656: Unknown event
> 161031660: engineUpdate, opmode=0x888
> 161031742: TXMODE, freq=923000000, len=20, SF=10, BW=125, CR=4/5, IH=0
> start single rx: now-rxtime: 4
> 161115135: RXMODE_SINGLE, freq=923000000, SF=10, BW=125, CR=4/5, IH=0
> rxtimeout: entry: 161123842 rxtime: 161115115 entry-rxtime: 8727 now-entry: 4 rxtime-txend: 60196
> start single rx: now-rxtime: 4
> 161174561: RXMODE_SINGLE, freq=923200000, SF=10, BW=125, CR=4/5, IH=0
> rxtimeout: entry: 161189414 rxtime: 161174543 entry-rxtime: 14871 now-entry: 4 rxtime-txend: 119624
> 161445288: EV_TXCOMPLETE (includes waiting for RX windows)
> RSSI: 25
> SNR: 45
> 161445298: engineUpdate, opmode=0x800
> 161757797: engineUpdate, opmode=0x808
> Packet queued
> 163348378: engineUpdate, opmode=0x8
> 163348457: Unknown event
> 163348462: engineUpdate, opmode=0x888
> 163348546: TXMODE, freq=923200000, len=20, SF=10, BW=125, CR=4/5, IH=0
> start single rx: now-rxtime: 4
> 163431940: RXMODE_SINGLE, freq=923200000, SF=10, BW=125, CR=4/5, IH=0
> rxtimeout: entry: 163440647 rxtime: 163431920 entry-rxtime: 8727 now-entry: 4 rxtime-txend: 60196
> start single rx: now-rxtime: 4
> 163491366: RXMODE_SINGLE, freq=923200000, SF=10, BW=125, CR=4/5, IH=0
> rxtimeout: entry: 163506219 rxtime: 163491348 entry-rxtime: 14871 now-entry: 4 rxtime-txend: 119624
> 163803437: EV_TXCOMPLETE (includes waiting for RX windows)
> RSSI: 25
> SNR: 45
> 163803447: engineUpdate, opmode=0x800
> 164115948: engineUpdate, opmode=0x808
> .
> .
> .
> .
> 198100321: engineUpdate, opmode=0x8
> 198100393: Unknown event
> 198100397: engineUpdate, opmode=0x888
> 198100481: TXMODE, freq=923400000, len=20, SF=10, BW=125, CR=4/5, IH=0
> start single rx: now-rxtime: 4
> 198183874: RXMODE_SINGLE, freq=923400000, SF=10, BW=125, CR=4/5, IH=0
> rxtimeout: entry: 198192581 rxtime: 198183854 entry-rxtime: 8727 now-entry: 4 rxtime-txend: 60196
> start single rx: now-rxtime: 4
> 198243300: RXMODE_SINGLE, freq=923200000, SF=10, BW=125, CR=4/5, IH=0
> rxtimeout: entry: 198258154 rxtime: 198243282 entry-rxtime: 14872 now-entry: 3 rxtime-txend: 119624
> 198517809: EV_TXCOMPLETE (includes waiting for RX windows)
> RSSI: 25
> SNR: 45
> 198517819: engineUpdate, opmode=0x800
> 198830319: engineUpdate, opmode=0x808
> Packet queued

terrillmoore, 2018-11-26:
Hi @3nd0y -- we'll have to investigate this. Which network are you using?  I found that I had to use `void LMIC_setDrTxpow (dr_t dr, s1_t txpow);` to set the data rate. (I recommend coding the txpower at 14dB because currently it's ignored but that's going to be fixed soon.) Or you can try ADR.

terrillmoore, 2018-11-26:
I looked at the code. Indeed, in v2.3.0 and earlier, you need to call LMIC_setDrTxpow() if you want any controlled data-rate after JOIN.  It certainly would be better to have a regional function that sets the initial post-join datarate to the fastest narrow-bandwidth rate available, as that's best for the network and for battery life; but it is somewhat application specific. I'll file a bug report on this.

By the way, if you use `KEEP_TXPOW` as the power parameter, the transmit power in `LMIC.adrTxPow` won't be changed.