mcci-catena/arduino-lmic #202: [Feature Request] option to use ESP32 AES hardware encryption

cyberman54, 2018-12-02:
ESP32 cpu provides AES hardware encryption.
The ESP32 SDK has two libraries: `aes.o` and `sha.o` providing several functions.
Using hwcrypto may speed up the LMIC library and saves storage (flash + ram) on ESP32 systems.

Perhaps could be done by wrapping
`void lmic_aes_encrypt(unsigned char *Data, unsigned char *Key)`

manuelbl, 2018-12-16:
In my library for the ESP-IDF framework, the hardware encryption is used. I assume (I never measured it) that it reduces the code sizes. I also believe that it is slower than a good software implementation – the reason being that I've gone for the simplest approach by just implementing *lmic_aes_encrypt*. So the key needs to be set for each AES block – not super efficient. But then the amount of data that's being encrypted is very small.

If you want to use it as well, you'll need the file [mbedtls_aes.c](https://github.com/manuelbl/ttn-esp32/blob/master/src/aes/mbedtls_aes.c) and you will need to modify *config.h* so it supports the preprocessor define *USE_MBEDTLS_AES*.

cyberman54, 2018-12-16:
Thanks for your note!
So, is it correcht that means i need two defines in `lmic_config.h` ?

```
#define USE_ORIGINAL_AES
#define USE_MBEDTLS_AES
```

manuelbl, 2018-12-16:
`USE_MBEDTLS_AES` must be defined, but **not** `USE_ORIGINAL_AES`. Furthermore [config.h](https://github.com/mcci-catena/arduino-lmic/blob/master/src/lmic/config.h#L152) must be modified such that it does not define `USE_IDEETRON_AES`.

terrillmoore, 2018-12-26:
@cyberman54, unfortunately, there are no good ways to do this with config.h, because it requires some configuration variable to be set. It would be sensible to allow this to be overridden by something external, as in "USE_EXTERNAL_AES" -- that would turn off both of the stock defines, and allow you to do your own thing.  Then you could put this in your `boards.txt` file for the Arduino environment.

cyberman54, 2019-05-25:
@terrillmoore Since Espressif announced the new ESP32-S2 with crypto feature, would be nive to use this feature with it.

https://www.espressif.com/en/news/espressif-announces-%E2%80%A8esp32-s2-secure-wi-fi-mcu

DylanGWork, 2021-07-12:
Hi, I'm having trouble getting this to work, getting the error:
../components/src/aes/other.c: In function 'os_aes_cmac':
../components/src/aes/other.c:56:9: error: implicit declaration of function 'lmic_aes_encrypt' [-Werror=implicit-function-declaration]
         lmic_aes_encrypt(AESaux, AESkey);
         ^~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors

I see this line but I'm not sure where it should be defined?

// This should be defined elsewhere
void lmic_aes_encrypt(u1_t *data, u1_t *key);

DylanGWork, 2021-07-13:
I ended up using the way it was done in this library and have it working:
https://github.com/TobleMiner/lmic-esp-idf/tree/master/src/aes

cyberman54, 2021-07-13:
@DylanGWork So, that means you changed code inside of the MCCI LMIC library? Can you make a PR?

DylanGWork, 2021-07-13:
Sure did, I've made a few alterations so I'll see if I can backtrack to a working version just using the edit for this AES stuff