mcci-catena/arduino-lmic #203: LMIC_setClockError - affects OTA?

cyberman54, 2018-12-20:
A user of my paxcounter software, which is based on MCCI LMIC, reports that `LMIC_setClockError` affects OTA joining - but in a way that smaller RX windows give better results. I don't understand this, would assume the opposite?

https://github.com/cyberman54/ESP32-Paxcounter/issues/225

using

`LMIC_setClockError(MAX_CLOCK_ERROR * CLOCK_ERROR_PROCENTAGE / 100);`

terrillmoore, 2018-12-26:
I would also assume the reverse. The window opens earlier when clock error is made greater; this means that there is more chance for noise to close the receive window. I guess that this might be the problem for your user. Unfortunately, I can't see from the printout exactly why and when the receive got aborted.  Perhaps we should pursue this by adding a different kind of debug printout to the LMIC code so we see high-level events (not things related to the radio interface wrangling). But not being in France, I can't easily repro. Both default and wider windows seem to work well with The Things Network for me.  What else is his software doing, by the way? On the ESP32, FreeRTOS is running in the background, right?

ngraziano, 2019-02-12:
Hi,

I think for big clock error (>10%) the timing on RX1 and RX2 on join request may overlap on slow datarate.
Exemple for  20% like in https://github.com/cyberman54/ESP32-Paxcounter/issues/225 you have the following delay after end of TX (optimal is 5s for RX1 and 6s for RX2).
Time in ms after TX end calculate by `schedRx12` (if haven't make mistake):

|    DR |  RX1 start | RX1 timeout | RX2 Start |
|-------|------------|--------------|------------|
|SF12 | 4049 |**6212** | 4853 |
|SF11 | 4025 |**6106** | 4828 |
|SF10 | 4012 |**6053** | 4988 |
|SF09 | 4494 |_5539_ | 5494 |
|SF08 | 4747 | 5269 | 5747 |
|SF07 | 4873 |5135 | 5873 |

In case of join on RX2 with a datarate < SF10 the two RX windows may overlap.

The full sheet I use to make the calculation: [Clock Error timing](https://docs.google.com/spreadsheets/d/1vAqOPp15W_rK54vlH9GMd-LzKTge6VHgcJRs5AJFJwc/edit?usp=sharing)

Maybe we can add a warning comment for clock error > 7% (where the problem may begin).

cyberman54, 2019-02-12:
That's a very reasonable explanation, thanks for putting effort on this. In my paxcounter software i will add compiler directive to check value of clock error. Maybe this can be done in LMIC code, too.

terrillmoore, 2019-02-12:
The whole clock-inaccuracy thing is a bit of a crock, in any case; the fundamental problem is that (in an Arduino env) it's very hard to schedule things accurately. With some recent changes, we can greatly improve this by using an ISR, getting the exact end-of-tx time in the ISR, then passing it into the code when we get around to it.  However this should be lower priority than port 224 support and getting us to the point of passing LoRaWAN compliance -- at that point we can build proper regression tests and we can be much more relaxed about making changes.