mcci-catena/arduino-lmic #207: LoRa gataway

Skodra, 2019-01-24:
There is no documentation for this library, there are Some examples, but not one that would show how to make LoRa Gateway, that is to make LoRa module be in constant state of listening to the incoming messages. Is this even possible with this library?

frazar, 2019-01-24:
For a Class A node, receiving a packet can only happen after a transmission. So, in order to be "constantly" listening, you can keep sending packets at a regular interval.

This is exactly what is implemented in the "ttn-otaa" example, where a "Hello, world!" message is sent every `TX_INTERVAL` seconds. You can read any incoming packet when an event of type `EV_TXCOMPLETE`. If a message is received, then
[ttn-otaa.ino.zip](https://github.com/mcci-catena/arduino-lmic/files/2791106/ttn-otaa.ino.zip)
[ttn-otaa.ino.zip](https://github.com/mcci-catena/arduino-lmic/files/2791107/ttn-otaa.ino.zip)

 `LMIC.dataLen` will be greater than zero, and the payload data can be found at the address `LMIC.frame + LMIC.dataBeg`.

I attach a slight modification of the "ttn-otaa" that prints every received message. I'm not able to test it right now, so let me know if it works.
[ttn-otaa.ino.zip](https://github.com/mcci-catena/arduino-lmic/files/2791111/ttn-otaa.ino.zip)

Skodra, 2019-01-24:
Thank you for your response and example. I am interested in making Class C node with this library, that is simple gateway, do you know is this possible?

Skodra, 2019-01-24:
OK, I found this in doc/LMiC-v1.5.pdf

> The IBM LoRaWAN C-library (LMiC) is a portable implementation of the LoRa™ MAC specification for
the C programming language. It supports both the EU-868 and the US-915 variants of the specification
and **it can handle class A and class B devices**.

So I can't use this library for building a gateway, does anybody know what is standard library for that?

terrillmoore, 2019-01-24:
It's possible, but not a target use case. This library is targeting small devices. For a gateway, you'd need a lot more code. You probably would be better off with the Radio Head library. (Search for that.)

Skodra, 2019-01-24:
But with the RadioHead library I would not be able to utilize LoRaWAN protocol, but only LoRa modulation. If someone knows the best way (library that is standard, that has documentation and examples) to make Class C (gataway) device out of lets say RFM96 please post it here.

terrillmoore, 2019-01-24:
This would be better raised at https://thethingsnetwork.org forums. I need to keep the issues focused on bug fixes. Thanks!

closed 2019-01-24 by terrillmoore