mcci-catena/arduino-lmic #216: bootloop on esp8266 esp-12e with dragino shield

aparcar, 2019-02-12:
Hi, flashing works fine but I'm stuck in a boot loop, receiving an error like this:

```
16:03:50.336 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
16:03:50.336 -> 
16:03:50.336 -> load 0x4010f000, len 1384, room 16 
16:03:50.336 -> tail 8
16:03:50.336 -> chksum 0x2d
16:03:50.336 -> csum 0x2d
16:03:50.336 -> v951aeffa
16:03:50.336 -> ~ld
```

Can't figure out whats the problem... any advise?

terrillmoore, 2019-02-13:
I'm sorry, not enough info. Put more debug prints in your sketch so you can see exactly where it's crashing.

aparcar, 2019-02-16:
Works now, I badly connected the miso cable...

closed 2019-02-16 by aparcar

terrillmoore, 2019-02-16:
Thanks for updating! Glad it's working!

aparcar, 2019-02-16:
Glad you keep this repo up to date :)!