mcci-catena/arduino-lmic #219: Usage with Arduino One and Dragino Lora Shield

aparcar, 2019-02-15:
I'm trying to use this library with an Arduino Uno and the Dragino Lora Shield, following the "official pinmap"

```
// Pin mapping<br>const lmic_pinmap lmic_pins = {
    .nss = 10,// Connected to pin D10
    .rxtx = LMIC_UNUSED_PIN,// For placeholder only, Do not connected on RFM92/RFM95
    .rst = 9,// Needed on RFM92/RFM95? (probably not)
    .dio = {2, 6, 7},// Specify pin numbers for DIO0, 1, 2
// connected to D2, D6, D7 
};
```

However it never sends anything. However, with the old IBM lib it works. Am I missing something?

terrillmoore, 2019-02-16:
I'm really not sure. We don't have hardware so can't test. Which sketch are you using, which region are you using, and what do you get as output?

aparcar, 2019-02-16:
I'm using [this](https://github.com/matthijskooijman/arduino-lmic/) library. Using your result in a warning to use an explicit pinmap, which I'm trying. However the chip isn't really activated. I'll do some further debugging and let you know. Thanks for the quick resonse

aparcar, 2019-02-16:
works, hat to reset the ardunio ide but works now just fine!

closed 2019-02-16 by aparcar