mcci-catena/arduino-lmic #220: How to Query available duty quota?

hb9eue, 2019-02-16:
Hi

Sorry if this question is simple, but I am a very beginner with C++ and Arduino.

I use an Heltec ESP32 SX1276 Lora Board as a TTNMapper.org Node.
I took the example from: https://www.thethingsnetwork.org/labs/story/heltec-lorawan-gps-quick-start and also included quite some features I found in the examples of this library. So far this works quite well.

But there is one problem that keeps nagging me:

What does exactly happen, when I call LMIC_setTxData2 but I am over the allowed duty cycle?

The examples let me assume, the TX is delayed until the duty on the frequency allows TX. But that would be very bad for a driving node. Between acquiring the GPS position and the actual TX, the node could have traveled quite far!

Running the library in with debug enabled, I seem to observer that TX is just suppressed when I am over duty.

So, before queuing a send job, I would like to query if I have available duty airtime. How can I do this?

If I try:

    Serial.print(F("Duty Avail: "));
    Serial.println(osticks2ms(LMIC.globalDutyAvail));
    Serial.print(F("Duty Rate: "));
    Serial.println(osticks2ms(LMIC.globalDutyRate));

I always get zero as value.

Any hints appreciated!

-Benoît-

terrillmoore, 2019-02-16:
There are a number of duty cycle measurements, not just the global. I presume that one of the channel duty cycle counters is causing a problem.

Are you getting a EV_TXCOMPLETE?  According to the API (and this behavior isn't tested: we need a test suite!), the LMIC flags at the time of TXCOMPLETE are supposed to give you an indication as to whether the transmit actually was done.  That said: I don't have complete faith in that, because my test code that submits an acked uplink always succeeds, even when not acked. I've not tracked that down, so I'm wary of recommending using the TX_COMPLETE status. Please look at the current docs, https://github.com/mcci-catena/arduino-lmic/blob/master/doc/LMIC-v2.3.pdf, which I updated based on looking at the code. If I got that wrong, we need to fix that, too!

Does the uplink count change when the TX is suppressed?

hb9eue, 2019-02-17:
I'll look at the flags after transmitting.
Not reaching TX_COMPLETE could be one other issue I am observing, but I as not sure if I made some other mistake.
The example code schedules the next send_job when the TX_COMPLETE from the previous one is called.
From time to time, my TTN-Mapper Node just stops operating after several hours. Of course I then never have a serial console connected to figure out why this happened.
So a scheduled sending job which never calls TX_COMPLETED would lead to exactly that observation.

I'll read the docs more thoroughly :-)

-Benoît-