mcci-catena/arduino-lmic #221: Opmode OP_TXRXPEND not being reset after send

hb9eue, 2019-02-16:
After updating from 2.3.1 to 2.3.2 my sketch does not send anymore.

Debug output let's me suspect, that OP_TXRXPEND is not being cleared anymore after TX is completed.

1316477: engineUpdate, opmode=0x4
1356407: engineUpdate, opmode=0x4
EV_TXSTART
1366540: engineUpdate, opmode=0x884
1366610: TXMODE, freq=868100000, len=23, SF=7, BW=125, CR=4/5, IH=0
start single rx: now-rxtime: 3
1683077: RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
EV_JOINED
netid: 19
devaddr: 26012062
artKey: XXX
nwkKey: XXX
1691101: engineUpdate, opmode=0x800
Do Send!
OP_TXRXPEND, not sending

-Benoît-

hb9eue, 2019-02-16:
Ok, this issue can be deleted. After restarting the Arduino IDE and recompiling the problem disappeared.

terrillmoore, 2019-02-16:
Great! Closing.

closed 2019-02-16 by terrillmoore