mcci-catena/arduino-lmic #222: How do I run non-LoRaWAN code in LMIC

schwiderke92, 2019-02-16:
may seem stupid the question, but how to make a tx Lora, only when some event happens? Is there any command that can be done like send ()?
For example: if sensor == 25 {
send ()
}

schwiderke92, 2019-02-16:
I recently implemented an RFID along with LMIC, and had to remove the os_runloop_once () from the void loop, because as the RFID was sharing the SPI with LMIC, then when I went into RFiD, LMIC expected an SPI command to sleep (error 545) and the return value was from RFID which locked the code. I ended up having to implement the functions of the functions outside the main loop.
I was using SS on different pins, but I believe the SS of the radio was not being turned off by the LMIC

terrillmoore, 2019-02-20:
Looks like there are two issues here. I'll try to come back later today with comments.

terrillmoore, 2019-02-22:
I have successfully done event-driven uplinks from the LMIC. 

1. the LMIC is not designed to dynamically share the SPI bus. We can fix this, but we need to prioritize port 224 (compliance) support, so we can do hardware-in-the-loop regression testing. You can share from outside the LMIC provided that you make sure NSS is inactive (high) whenever you're outside the LMIC. A wrapper function around os_runloop_once() should suffice.

2. I didn't follow the docs when I wrote my code (I wasn't aware that there's a suggested pattern).   My `loop()` function was:

```c
void loop() {
   os_runloop_once();
   SensorFSM_eval();
   LoRaFSM_eval();                                                                     }
}
```

As the names imply, I have two finite state machines; one for the sensor, one for the radio transmission.

The reason for the FSM for LoRa is to deal with sensor events that occur while you're already busy doing something. Otherwise things will get stuck or confused.

I will try to convert my code to a suitable example application, but it may take some time.

schwiderke92, 2019-03-01:
Thanks for help!