mcci-catena/arduino-lmic #224: Handle "NOT_A_PIN" in Arduino_pinmap.h

cyberman54, 2019-02-19:
For unused but connected pins Arduino-esp32 core is using a macro `NOT_A_PIN `in it's board files, which resolves to -1, while LMIC is using "LMIC_UNUSED_PIN". Using Arduino board files with LMIC this will cause a compiler error evaluating the .pConfig struct:

Example:

Given Arduino board file...
`https://github.com/espressif/arduino-esp32/blob/master/variants/lopy4/pins_arduino.h`

...containing
`#define LORA_RST NOT_A_PIN`

...resolves to -1 in arduino-esp32 framework
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Arduino.h#L115

...results to:
```
error: narrowing conversion of '-1' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
.pConfig = &myHalConfig};
```

terrillmoore, 2019-02-19:
This is tough; it can't be fixed on our side. LMIC API requires uint8_t here. 

Since LORA_RST is not defined by us, when defining a pinmap, the coder has to use it anyway, and so can write:

```c++
uint8_t(LORA_RST == NOT_A_PIN ? LORA_RST :  LMIC_UNUSED_PIN)
```

in the places where `LORA_RST` is used.  Feel free to reopen if I've missed something.

cyberman54, 2019-02-19:
Thanks for your suggestion to do the mapping in the application, i did not have that idea - easy patch, did it, ready.

closed 2019-02-19 by cyberman54