mcci-catena/arduino-lmic #226: CFG_us915 enabled by default

aapris, 2019-02-22:
[README.md](https://github.com/mcci-catena/arduino-lmic/blob/master/README.md#selecting-the-lorawan-region-configuration) says:

> You should define exactly one of CFG_... variables. If you don't, the library assumes CFG_eu868.

However, in [lmic_project_config.h](https://github.com/mcci-catena/arduino-lmic/blob/d112c8606a70ef6f7282a79d91995d075b996e30/project_config/lmic_project_config.h#L3) there is line 
`#define CFG_us915 1`
commented in.

If I comment this out and enable e.g. `CFG_eu868`, this will get overwritten in next library update if Arduino IDE is used. This is bad, because after library update code still compiles fine, but frequency has changed back to `us915` without any notification.

I'd say all of those configs should be commented out and compilation should fail, if developer hasn't explicitly commented in exactly one of those configuration options.

terrillmoore, 2019-02-22:
Unfortunately, this is a limitation in the Arduino environment and cannot be fixed without a change to the environment: you can't push info from your sketch into the library compile environment. This is a common issue with all the LMIC variants (although others use EU as the default). It is *not* a problem if not using the Arduino environment because you can trivially add -D; and this code is used in non-Arduino environments.

You may look at how MCCI handles this in our BSPs, or how @frankleonrose handles this using PlatformIo.  In MCCI's BSPs, we add a menu entry to the Arduino environment, so you select your region from the IDE. Frank uses PlatformIO to do a `-D` when compiling.

Sorry!

closed 2019-02-22 by terrillmoore

lnlp, 2019-08-08:
The biggest problem with settings that get undone/overwritten by library updates is that the user is not notified when this has happened and the code will compile successfully but fail to run successfully without any clear indication because the region setting (a.k.a. band plan) was overwritten.

A fix has been proposed in #358. 
The proposed fix will at least notify the user that the region setting was overwritten (by notifying that a region has not yet been selected/set).
It also makes clear to every user that the correct region has to be explicitly selected/set by the user  (no default).