mcci-catena/arduino-lmic #227: LoraWAN 1.1 security support at some point?

pomplesiegel, 2019-02-25:
Hello, 

This is a potentially off-putting question, but I mean no disrespect: Is there a plan to eventually integrate LoraWAN v1.1 security support into LMIC? 

LMIC is incredible and I am eternally grateful for the community surrounding it. We're having a blast using it. 

Thank you!

terrillmoore, 2019-02-25:
Thanks for the kind words. 

Yes, definitely the plan is to upgrade to add LoRaWAN v1.1 features, including: 

1. All the new join features.
2. Multicast
3. Class B and Class C
4. Hooks for FUOTA

However, we first have to make LMIC testable, so we are sure we are not breaking things as we add new features. So #215 is going to block progress. Luckily, #215 doesn't involve any changes to the LMIC until we fix all the bugs that are revealed. Once the bugs are fixed, we can move forward to 1.1.  If we can get some automation going with regression testing, we can probably open up development. MCCI has a RedwoodComm [RWC5020A](http://redwoodcomm.com/product/01.php?cate_1=32) tester, so we should be able to test port 224 support, and possibly even get to the point of HW in the loop testing. 

I'm going to edit the issue title.

pomplesiegel, 2019-02-25:
Great, thank you for the detailed answer! Looking forward to all of this

mrx23dot, 2022-02-28:
Is it backward compatible with v1.0.x providers?

terrillmoore, 2022-02-28:
v1.1 the spec is backward compatible if implemented properly; our implementation doesn't exist, so this doesn't (yet) apply. However, there's not a lot of demand for v1.1 -- other things are more important. Definitely less works to implement v1.0.4, but even that is of less value than class C under v1.0.3.  v1.0.3 will be around forever in the network providers.