mcci-catena/arduino-lmic #228: Add PlatformIO support

terrillmoore, 2019-02-25:
We need the usual json files for using the library with PlatformIO. Also need CI tests (which will run much faster than the Arduino tests, and maybe can be used for primary breadth-coverage). Since I don't use PlatformIO much, help would be appreciated.

frazar, 2019-02-25:
I am currently using this library with PlatformIO so, in some sense, PlatformIO is already supported.

I might help with setting up the CI. Would compiling the example sketches be sufficient?

illperipherals, 2019-02-28:
I am using this within platformio as well. 
[image]
I can assist with tasks.