mcci-catena/arduino-lmic #231: Documentation is scattered

frazar, 2019-02-26:
### Disclaimer

I will gather some personal thoughts on the state of documentation of the library. I am aware that I might be the only one with this point of view. If that's the case I'll close this issue.

### The problems
Right now, the documentation is scattered in different places:
- the .docx file in the doc folder: inherited from the original IBM project?
- the README.md file, which is rather long and comprehensive
- the GitHub wiki: which comprises a couple of pages
- the comments in the code

Moreover, the main documentation is currently stored in a docx format and exported in pdf. format I find there are a few drawbacks to this approach:
- it is not possible to check and review edits to the docx files through the online GitHub interface
- editing docx requires having an Office suite 
- editing docx with different Office suites (Micorsoft Office, LibreOffice, OpenOffice, etc) might generate inconsistencies
- the genrated pdf files are binary and large, and git stores them as a whole, increasing the size of the repo

### The solution
Ideally, there should be just **one place** where documentation is concentrated. In addition to a concise and terse README.md pointing out the most basic things (scope of the projects, latest release version, a link to documentation source, release policy, link to contribution guidelines).

For the main documentation, I suggest we consider some of the available solutions to software documnetation:
- [Doxygen](http://www.doxygen.nl/index.html): popular for C++ projects, automatically generates a documentation website from annotaitons in the code. This helps ensure that code and documentation are coupled together. An [example](https://dbus.freedesktop.org/doc/api/html/annotated.html) page of the documentation generated by Doxygen.
- [ReadTheDocs](https://readthedocs.org/): popular among Python projects, free hosting provided, with appealing graphics. An [example](https://numpy.readthedocs.io/en/stable/reference/routines.financial.html) page of the documentation generated by ReadTheDocs.
- GitHub wiki: already in use
- others?

### Summary
Summarizing, I propose
- documentation should be kept in a single place, using appealing tools that fit the needs of Arduino LMIC
- README should be cut to the minimum

I think consistent documentation is what distinguishes high quality libraries. I realize this might be a lot of work, but I am willing to personally take care of the migration process once we agree on the way to go.

terrillmoore, 2019-03-01:
I agree with the general comments. I don't like doxygen much, because it encourages a worms-eye view, and ... it looks ugly to me, both in the presentation and in the code. (I use the libusb docs all the time, and I am constantly jarred by this.)  Markdown is too weak for presenting reference info. However, a separate Word doc just doesn't get read or updated. Let me reflect on this. I think that the port 224 this is the burning need at the moment.

frazar, 2019-03-01:
Yes, I agree that the default Doxygen style is not very appealing. Moreover, if only documentation describing the available methods is produced, it can lead to the worms-eye view you mentioned. 

However, there are a few ways to customize Doxygen's output, depending on the amount of work one wishes to invest. See [here](https://github.com/Velron/doxygen-bootstrapped). Moreover, Doxygen allows for richer documentation, like [the very nice doc](http://eigen.tuxfamily.org/dox/) for the Eigen library.

Finally, I expect the average LMIC contributor, as a C/C++ developer, would be more familiary with Doxygen than with other solutions.