mcci-catena/arduino-lmic #248: CFG_eu868 not working

WhiteTowerBadger, 2019-03-13:
I use lmic_project_config.h mentioned below together with an adafruit lora feather M0 900Mhz, but it still transmits +900 MHz and not in the 868 band as required. 
```
// project-specific definitions
#define CFG_eu868 1
#define CFG_sx1276_radio 1
//#define LMIC_USE_INTERRUPTS```

terrillmoore, 2019-03-13:
Not sure what to say; it’s working for others. Can you be more specific about your environment? 

 

If you’re using the standard Arduino IDE, can you first try hiding all your ‘libraries’ directories (rename the dir to libraries.hide) and then install the Arduino-lmic library from scratch? It’s most likely that another copy of the library is in your setup. Alternately, PlatformIO (because of how it handles libraries) might be a good test.

 

Alternately, please try installing MCCI’s BSP (https://github.com/mcci-catena/arduino-boards) , selecting ‘Catena 4410’ as the target board (that’s a Feather M0 LoRa, the extra things are not relevant here), and then select EU868 from the Arduino UI? That will force CFG_eu868 on the compiler command line, and will likely make the problem easier to find.

 

--Terry

terrillmoore, 2019-05-20:
I'm closing this because others are using this successfully.

closed 2019-05-20 by terrillmoore

porkyneal, 2019-06-18:
Sorry to re-open this issue, but I too am having issues using the library with EU config.

I am using platformio adding the following to my build flags: 

```-DCFG_eu868 -DLMIC_DEBUG_LEVEL=2 -DLMIC_PRINTF_TO=Serial```

This is giving a compiler warning:

 ```error You can define at most one of CFG_... variables error You can define at most one of CFG_... variables```

I can compile by replacing "-D CFG_eu868" with "-D LMIC_REGION_eu868" however this doesn't appear to do the job, with the debug showing it is TX on frequencies in the 900MHz range.
```
Starting
RXMODE_RSSI
10854: engineUpdate, opmode=0x808
Packet queued
10884: EV_JOINING
10888: engineUpdate, opmode=0xc
10920: EV_TXSTART
10980: TXMODE, freq=909500000, len=23, SF=10, BW=125, CR=4/5, IH=0
330726: setupRx1 txrxFlags 00 --> 01
start single rx: now-rxtime: 3
330860: RXMODE_SINGLE, freq=925700000, SF=10, BW=500, CR=4/5, IH=0
rxtimeout: entry: 363501 rxtime: 330850 entry-rxtime: 32651 now-entry: 4 rxtime-txend: 296692
372425: setupRx2 txrxFlags 0x1 --> 02
start single rx: now-rxtime: 3
```

Am I missing something obvious?

porkyneal, 2019-06-18:
OK So i added my comments slightly premature, for all those struggling:

terrillmoore, 2019-06-18:
Yes, `-D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS` is required for Platform I/O.

jipp, 2021-10-17:
HI

sorry to jump back on this issue.

I use Arduino IDE 1.8.16 on osx
only lib installed is: `https://github.com/mcci-catena/arduino-lmic`
using example: `ttn-otaa-feather-us915`

running from plain installation everything compiles.

now I modify the lmic_project_config.h:
```
diff lmic_project_config.h lmic_project_config.h.old 
2,3c2,3
< #define CFG_eu868 1
< //#define CFG_us915 1
---
> //#define CFG_eu868 1
> #define CFG_us915 1
```
after this modification I see the following in the IDE console:

```
Linking everything together...
/Users/jipp/Library/Arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-g++ -L/var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066 -Os -Wl,--gc-sections -save-temps -T/Users/jipp/Library/Arduino15/packages/adafruit/hardware/samd/1.7.5/variants/feather_m0/linker_scripts/gcc/flash_with_bootloader.ld -Wl,-Map,/var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/ttn-otaa-feather-us915.ino.map --specs=nano.specs --specs=nosys.specs -mcpu=cortex-m0plus -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/ttn-otaa-feather-us915.ino.elf /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/sketch/ttn-otaa-feather-us915.ino.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/aes/lmic.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/aes/other.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/aes/ideetron/AES-128_V10.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4420.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4551.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4610.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4611.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4612.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4617.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4618.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4630.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4801.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_catena4802.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_disco_l072cs_lrwan1.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_feather32u4lora.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_featherm0lora.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_heltec_lora32.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_thisboard.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_ttgo_lora32_v1.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/getpinmap_ttgo_lora32_v2.1.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/hal/hal.cpp.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/lmic/lmic.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/lmic/lmic_as923.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/lmic/lmic_au915.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/lmic/lmic_channelshuffle.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/lmic/lmic_compliance.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/lmic/lmic_eu868.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/lmic/lmic_eu_like.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/lmic/lmic_in866.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_library/lmic/lmic_kr920.c.o /var/folders/54/74wflpt92tscg4xhfbq_lm340000gp/T/arduino_build_827066/libraries/MCCI_LoRaWAN_LMIC_ [... cut ...]
Bibliothek MCCI_LoRaWAN_LMIC_library in Version 4.1.0 im Ordner: /Users/jipp/Documents/Arduino/libraries/MCCI_LoRaWAN_LMIC_library  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: /Users/jipp/Library/Arduino15/packages/adafruit/hardware/samd/1.7.5/libraries/SPI  wird verwendet
Bibliothek Adafruit_ZeroDMA in Version 1.1.0 im Ordner: /Users/jipp/Library/Arduino15/packages/adafruit/hardware/samd/1.7.5/libraries/Adafruit_ZeroDMA  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit Feather M0.
```

what would be the recommended way to modify the frequency to eu868?

thx in advance