mcci-catena/arduino-lmic #255: Add EV_RXSTART and EV_JOIN_TXCOMPLETE

terrillmoore, 2019-03-18:
When debugging compliance testing, I found I needed these in order to properly log what was going on. EV_RXSTART is a tricky one, as it is issued when receive is about to start; the caller had better not do anything time consuming. So don't pass it to old `onEvent()`. On the other hand, `EV_JOIN_TXCOMPLETE` just indicates that a join transmit cycle has completed without success, and it's not time-critical.

terrillmoore, 2019-03-18:
Committing as part of #215.

closed 2019-03-19 by terrillmoore

jezdd, 2019-03-19:
Don't these need adding to example code too? I'm seeing 'Unknown event' otherwise.

terrillmoore, 2019-03-19:
Thanks for your comment. Yes. I want to pull the event name table into the API, so we can add events as needed; see the compliance app.

 

This is still all pre-release (as indicated by a release tag), but we need to get people testing.

 

BTW: I found, during test, that printing in the event routine is very bad for the LMIC’s health. (It’s also not good to do something “eagerly” when you get a notification from the network.)  Some of the events are called at time critical moments. It’s not a big deal during uplink, but when there is downlink traffic (either for ADR, or for confirmed uplink, or because the application needs downlink), the prints are really troublesome. Check the compliance example to see how I’m doing it there. That all needs to go into the Arduino-LMIC framework, so that novices don’t have to deal with this. So all the examples have to change, and probably in a pretty big way.

commit 1b2db4e8 (2019-03-18): Fix #255: add EV_RXSTART and EV_JOIN_TXCOMPLETE