mcci-catena/arduino-lmic #265: raw-feather example doesn't compile under PlatformIO

terrillmoore, 2019-03-18:
> I'm not sure how to correctly solve this, but commenting the `extern "C"` it compiles fine.

Unfortunately, this looks like a defect in PlatformIO's scanning of .ino sketches. The `extern "C"` is intentional. Apparently PlatformIO is lifting the non-`extern "C"` definition to the synthesized header file that wraps the .ino, and missing the `extern "C"` part. We may have to omit raw-feather from the list of apps, because this kind of problem is really hard to fix. Or one of us can take the problem to @ivankavets, he's very responsive.  (Since this compiles fine with Arduino, and Ivan has his own .ino to .cpp converter, I'm guessing that this is the issues. I've seen this kind of thing before with Visual Micro, which also has it's own .ino parser.)

_Originally posted by @terrillmoore in https://github.com/mcci-catena/arduino-lmic/pull/236#issuecomment-473933692_

ivankravets, 2019-03-20:
Hi @terrillmoore, thank you so much for the mention!  This our bug with INO to CPP preprocessor. I've just fixed it https://github.com/platformio/platformio-core/commit/d9e6111ac3b19721a3c0ee9bb8cc3d5a6add4c44

The final release of PIO Core 3.6.6 is planned for the next week.

You can close this issue.

terrillmoore, 2019-03-20:
Thanks, @ivankravets!

closed 2019-03-20 by terrillmoore