mcci-catena/arduino-lmic #268: Compile error when option #define LMIC_ENABLE_DeviceTimeReq is used

cyberman54, 2019-03-19:
With commit `43caadd` If using option `#define LMIC_ENABLE_DeviceTimeReq` compiler runs on errors due to two missing definitions in `struct lmic_t` :

> error: 'struct lmic_t' has no member named 'pNetworkTimeCb'
> error: 'struct lmic_t' has no member named 'pNetworkTimeUserData
> LMIC.pNetworkTimeCb = pCallbackfn;
> LMIC.pNetworkTimeUserData = pUserData;

After disabling the option code compiles fine.
I guess this will be a simple fix, but did not yet found time to peek into the code changes.

terrillmoore, 2019-03-19:
I hate `#if`. We don't have CI testing that covers this case. However, just to confirm, it's *not* `#define LMIC_Enable_DeviceTimeReq`, it's `#define LMIC_Enable_DeviceTimeReq 1` (or zero). Blank is not permitted, we don't use `#ifdef` for new options, we use explicit compares. The compiler silently treats blanks as zero.  It is (with difficulty) possible to check for a blank definition in config.h and report an error. I guess I'd better think about this.

Are you using `#define LMIC_ENABLE_DeviceTimeReq` or `LMIC_ENABLE_DeviceTimeReq  1`?

cyberman54, 2019-03-19:
Using
`#define LMIC_ENABLE_DeviceTimeReq 1`
in lmic_config.h, 

> HEAD is now at 43caadd Merge pull request #260 from mcci-catena/issue215

terrillmoore, 2019-03-19:
OK, thanks. I duplicated this. In fixing #246, I moved these fields to the "client data that survives LMIC reset". Unfortunately, I didn't test compile.

There's a bigger issue, @frazar and I probably need to discuss how to handle network time callbacks during reset. Right now, probably it's not enough, we should cancel the outstanding request. I'll post a fix shortly for this one, anyway.

cyberman54, 2019-03-19:
Regarding time sync: in my application code i realized a time sync handshake loop over app server, not using MAC DevTimeReq/Ans, since TTN v2 stack does not support this. It works, i achieve precision around 1 second, still to be optimized. I found out that for this purpose using timestamp Lmic.Txend is better choice. Because it does not include small random time (as forced by LoRaWAN spec) before opening RX windows. So  using Lmic.Txend should in theory give slightly more precision for a time sync handshake, than taking  timestamp after RX windows finished (as the code by @frazar does, as far as i see).

terrillmoore, 2019-03-19:
I agree, the primary time reference for any network time operation should be `LMIC.txend`, as it's the reference (apart from speed-of-light issues) used by the network for time-stamping uplink traffic.

closed 2019-03-19 by terrillmoore

cyberman54, 2019-03-19:
@terrillmoore Tested commit `e30e56e` -> works! Thanks!

-- 

A question, not regarding this issue, but useful for timesync handshake design:
Does `LMIC_sendAlive()` open up a receive window, so would be a lightweight option to force receive data with class A devices, i.e. a time_sync_answer?

omaranw, 2019-08-01:
> I found out that for this purpose using timestamp Lmic.Txend is better choice.

Can you explain this a bit? As per my understanding,  Lmic.Txend just returns the os_gettime from when Tx ended. This time seems to have nothing to do with the network time. I understand it will work great for timestamp as long as system never restarts. But once it does, the time stamps will reset to beginning and follow the similar pattern as last time, if not exactly the generate the same old values.

commit 1d978e74 (2019-03-19): Fix #268: these callbacks are now in client data
commit e30e56ec (2019-03-19): Merge pull request #273 from mcci-catena/issue268

Fix #268: these callbacks are now in client data