mcci-catena/arduino-lmic #269: LMIC_reset doesn't really deal with pending time callback

terrillmoore, 2019-03-19:
When I fixed #246, I made sure that time callbacks don't silently disappear when the LMIC code decides it needs to do an `LMIC_reset()`, as at frame counter overflow.

I didn't think this through. The network time callback probably should be canceled, just as we do for pending transmits #247.  @frazar is the contributor here, and so I would like his input on the appropriate action.

frazar, 2019-03-20:
Yes, I think the easiest way to adress this is to cancel the request.
Then `LMIC_reset()` should take care of notifying the user that the request did not succeeed.
Something along the lines of

```C++
#if LMIC_ENABLE_DeviceTimeReq
if (LMIC.txDeviceTimeReqState != lmic_RequestTimeState_idle) {
    lmic_request_network_time_cb_t *const pNetworkTimeCb = LMIC.client.pNetworkTimeCb;
        
    LMIC.txDeviceTimeReqState = lmic_RequestTimeState_idle;

    if (LMIC.client.pNetworkTimeCb != NULL) {
        // reset the callback, so that the user's routine
        // can post another request if desired.
        LMIC.client.pNetworkTimeCb = NULL;

        // call the user's notification routine.
        (*LMIC.client.pNetworkTimeCb)(LMIC.client.pNetworkTimeUserData, 0);
    }
}
#endif // LMIC_ENABLE_DeviceTimeReq
```

terrillmoore, 2019-03-20:
ok, since I'm hacking so heavily on this, I'll look at this during my next coding session (this weekend, I suppose). I also should do this when doing an unjoin.