mcci-catena/arduino-lmic #275: Recent change in ADR handling in tnn may generate downlink loop

ngraziano, 2019-03-21:
Change https://github.com/TheThingsNetwork/ttn/commit/d9310f89a8abef5fa42a3d339eb5c2e534f27a3a in TTN cause the network to send LinkADRReq each time the network receive ADRACKReq even if the ADR parameter do not change (before it was only on change).

LMIC schedule an ADRACKReq after each time it receive an LinkADRReq https://github.com/mcci-catena/arduino-lmic/blob/master/src/lmic/lmic.c#L704 (even if there is no change)

So it loop between LinkADRReq  and ADRACKReq  and each uplink may be followed by a downlink.

I see 3 option 
- Remove `LMIC.adrChanged = 1;` and wait ADR_ACK_LIMIT + ADR_ACK_DELAY  to detect if there is a loss of connection with the new parameter (the network should have send good parameter).
- Only set `LMIC.adrChanged = 1;` if there is a change in parameters.
- Kindly ask TTN not to send LinkADRReq  if parameter do not change and are accepted by the device.

Which solution do you prefer?

terrillmoore, 2019-03-23:
I'd say "only set `LMIC.adrChanged =1` if there's a change. However, we're close to being ready to test ADR with the RWC tester, I may have additional input.

pomplesiegel, 2019-03-26:
In my current use of TTN I'm seeing downlink messages after every single uplink transmission when ADR is enabled. Is this what you're referring to?

Thank you!

pomplesiegel, 2019-03-27:
@ngraziano, is this what you're describing?

drahdiwaberl, 2019-03-27:
might make sens linking this
https://www.thethingsnetwork.org/forum/t/recent-change-in-adr-handling-in-tnn-and-lmic-generate-downlink-loop/24164

terrillmoore, 2019-03-27:
I guess I'd better look at this soon. I wonder if this is related to all 5 of 6 devices that were using more recent LMIC failing during the US network outage -- all came back with frame counter zero. In some cases, ADR causes a rejoin, and that would explain things. I still have to recover the devices and examine them; the one at hand could only be recovered by forcing yet another join.

drahdiwaberl, 2019-03-27:
well, gotta add:
just opened the ttn console to monitor traffic of the gateway which i still have one of these ADR-nodes receiving - and lo and behold, no downlink since 4 uplink-packets (SF7) .. might something have changed in ttn's handling of this yet again?

terrillmoore, 2019-03-27:
> Only set `LMIC.adrChanged = 1;` if there is a change in parameters.

It's not to hard to accomplish this (once I fix the bug in how we handle channel maps).

1. Change `setDrTxpow()` to `bit_t` from `void`, and return `0` for no-change, non-zero for change. 

2. Fix #278

3. Add a new function that takes the same parameters as the relevant ones from print:

   https://github.com/mcci-catena/arduino-lmic/blob/6e5ebbeee75647dd27aed90175272ac6d30a06dd/src/lmic/lmic.c#L694-L695

   (namely `p1, chmap, chpage, uprpt`), and applies the changes by calling the various routines, and tracks whether there are any changes.

4. Use the info from step 3 to set adrChanged to zero at line 704 if successful and nothing has changed.

terrillmoore, 2019-03-27:
Unluckily, I'm almost out of time to work on this today. I won't be able to look at this before tomorrow afternoon. I'll try to do it then, and then run the ADR test from the RWC 5020A. (Maybe check EU band as well, just to be sure.) I'll start by committing on a branch so that others can perhaps update and test with live networks.

terrillmoore, 2019-04-01:
Actually, I'm looking at this now. The code is confusing, because: LMIC.adrChanged is only consulted one place:

https://github.com/mcci-catena/arduino-lmic/blob/3c20fe10c9da2f787f6c901e5aa0b6a9e0d06102/src/lmic/lmic.c#L1470-L1475

As far as I can tell, this does two things:

1. **It enables ADR, even if it was disabled!** This is definitely not good.
2. It causes the ADR FSM to switch to the state where it wants to ask the network for ADR info on the next uplink.

This enabling ADR even if disabled is bad. I've got that patched with the other changes for #278. That's the real root cause of this problem that I saw with my devices, because it triggered a rejoin that wasn't expected, and the OP_REJOIN stuff is also broken.  While fixing all this, I'm also fixing #276. If JOIN is disabled, then the LMIC will not attempt to rejoin, it will just go on. I've also corrected logic that caused EV_LINK_DEAD to be reported multiple times, which seems silly. Once is enough.

terrillmoore, 2019-04-01:
The reason the OP_REJOIN stuff is broken is that it goes back to zero and has to count to 32 again before rejoining. For devices that send only three times a day, that is over a week offline.  Triggering a full join, at least with the current lack of join backoff, will kill batteries. So adjusted the counts to send a rejoin after every uplink (once we decide things are dead).

terrillmoore, 2019-04-01:
I've pushed my fixes to branch https://github.com/mcci-catena/arduino-lmic/tree/issue275. Will test with the RWC 5020A tomorrow. May have to make a test script that artificially advances ADR state to test this. Unfortunately for me, the LMIC now works *so* well that whenever the network sends a LinkAdrRequest it will override my applications choice of data-rate and TX power. That needs to be incorporated into the logic for checking the parameters. I can similarly see the uplink-repeat being something I'd not like to accept in some circumstances. Ditto ridiculously low power if I know the device is mobile.  But the channel mask is something that should always be acceptable, no need to give client a vote.

terrillmoore, 2019-04-01:
The branch builds cleanly, so ... feel free to test.

pomplesiegel, 2019-04-01:
Great, i'll give this a shot today. Thank you for your hard work!

pomplesiegel, 2019-04-01:
Working well for me so far! The ADR seems to be back to normal. However strangely the downlink ADR change messages are not showing up under the "gateway" tab on TTN console for my Multitech Conduit. I'm not sure if that's related. 

They show up under my application, however, and they are going through, based on the observed SF changes.

terrillmoore, 2019-04-02:
Also works for me. The downlink ADR changes have never showed up for me in the console; don't know why, apart from the empty downlink entry. I'm going to file another issue for my "unlucky" problem above, but that doesn't prevent merging, so expect a pull request/merge soon. Review is welcome, even after I merge.

pomplesiegel, 2019-04-02:
Sounds good! Thank you

closed 2019-04-02 by terrillmoore

ngraziano, 2019-04-02:
Thank you for the fix. Sorry I did not have the time to track this issue more.

commit ea775f78 (2019-04-02): Fix #275: Fix #276: Fix #278: Fix #165: ADR probs
commit f78c6b46 (2019-04-02): Merge pull request #289 from mcci-catena/issue275

Fix #275: make ADR work properly
commit 474c44a5 (2019-04-03): Fix #275: Forgot to set LMIC.adrChanged correctly