mcci-catena/arduino-lmic #279: Wrong PIN mapping in example

SmithyAT, 2019-03-30:
For the Adafruit Feather 32u4, there is a wrong pin mapping in the examples and therefor RX is not working. 

`#elif defined(ARDUINO_AVR_FEATHER32U4)`

You have to change this line to this mapping:
`.dio = {7, 6, LMIC_UNUSED_PIN},`

Additionally, solder wire as described on the Adafruit website from DIO1 to D6.

terrillmoore, 2019-03-30:
Really?!  You are right, according to the schematic, they wire IRQ to D7, not D3 as on the M0 LoRa. I would not have suspected that. D7 doesn't come to a pin on the AVR32, but D3 does; whereas neither D3 nor D7 come to pins on the Feather M0. Oh well. Sorry, I should have checked more closely when I added this. Thanks for pointing this out.

SmithyAT, 2019-03-30:
No problem at all. I was just wondering why I never got a JOINED message and then started to look deeper but haven't checked the pin mapping. Later I remembered that there is a difference on the Adafruit website and bingo, the sketch is working.

commit 9f92312b (2019-03-31): Fix #279: Adafruit 32u4 LoRa uses D7 for IRQ
commit d95271bd (2019-03-31): Merge pull request #283 from mcci-catena/issue279

Fix #279: wrong pinmap entries for Adafruit 32u4 Lora
closed 2019-03-31 by terrillmoore