mcci-catena/arduino-lmic #290: Fix examples not to disable LinkCheck mode

terrillmoore, 2019-04-02:
Many (all?) of the examples call `LMIC_setLinkCheckMode(0)` during init. This is now a bad example. Clean this up.

bertrik, 2019-04-18:
This looks like an easy fix to do.

In fact I tried this in my code and it seems to work. However I do see a bit of unwanted behaviour:
* my uplink to the closest gateway appears to be a better connection than the downlink
* at some point the node (LMIC) thinks that the link is dead and starts a rejoin
* the TTN receives the join message, creates a new device address and sends a join accept, but the accept is not received on the device side.
* the device keeps sending uplink messages using the old device address, which now seem to get ignored.
* If the device wouldn't have tried a rejoin, the messages would still have come though, I think.

I realise this is not really a library problem. I'm not very familiar with the LoRaWAN protocol specification yet. Perhaps just a link quality problem (I simply should get a better antenna, better antenna location, orientation, get closer to the gateway, etc.). What do you think?

terrillmoore, 2019-04-18:
Were you testing with HEAD? That looks like the V2.3.2 behavior -- which I also ran into. The rejoin behavior is different in HEAD. (But of course, it might be broken there, too.)

bertrik, 2019-04-18:
This was with the v2.3.2, I will try the github version tonight!

bertrik, 2019-04-19:
I tried this over the past night. I started an OTAA join last evening, and this morning I saw a lot of failed attempts this morning at trying to rejoin and the node data not coming through anymore. A restart of the join procedure (LMIC_reset() and LMIC_startJoining()) triggered a successful OTAA join. I did not take logs. Next time, I'll record more logging