mcci-catena/arduino-lmic #297: Latest commits on Master branch break ADR on my test node using TTN

pomplesiegel, 2019-04-03:
My ADR seemed to be working fine as of the commits from April 1st, but the latest commits on April 2nd seem to cause it to stop working. 
I am seeing downlink messages transmitted in my application (using the application log window on TTN), but my node remains at SF10, even with ADR turned on:
```cpp
    LMIC_setAdrMode(true);
```

[image]

terrillmoore, 2019-04-03:
Thank you for testing. I guess I'd better set up a European gateway for testing here. 

From your trace, I can see that we're not setting ADRAckRequest in the header (because there's no port-zero downlink). And I don't see a downlink for LinkADRReq in the trace. Are you seeing those (in a different trace)?

One thing I did when debugging that helped a lot. After TXCOMPLETE, I checked to see if a port-0 message was received, and printed out LMIC.frame[0] to LMIC.frame[LMIC.dataBeg - 1]. Then you can see the downlink MAC messages.

LMIC_setAdrMode(true/false) won't do much for us here; it turns on the FCtrl.ADR bit, which tells the network that we want ADR; but that's the default, and we turn that on anyway.

Thanks,
--Terry

pomplesiegel, 2019-04-03:
Hello @terrillmoore, happy to test! FYI I'm here in Boston using 915Mhz. 

I do see the downlink messages after every uplink in the application window: 
[image]

However, as I mentioned on another issue, recently I haven't seen these downlinks in the gateway window, whether the ADR is working or not. 

I'll look into printing the downlink messages now. Thank you!

terrillmoore, 2019-04-03:
That is really weird. I think you have an intermediate commit. Have you updated this morning? That looks like yesterday's behavior.  I'm at 2c169ecef8da2803197ed5da282a2beaf90662ef. 

What sketch are you using to test, and what other libraries?

Thanks!

terrillmoore, 2019-04-03:
OK, I duplicated this with the ttn-otaa-halconfig-us915 app. I think it's because the example scripts all have `LMIC.setLinkCheckMode(0)`, which seems to cause a problem. Investigating.

pomplesiegel, 2019-04-03:
OK, glad i'm not crazy! Just did a clean build and saw the issue persist. Thank you for the hard work!

pomplesiegel, 2019-04-03:
I just enabled link check mode via 

```cpp
LMIC.setLinkCheckMode(true)
```
and the issue persists on my node, FYI. Perhaps I'll leave it on in the future though, as it sounds like a handy feature.

terrillmoore, 2019-04-03:
I added this in TXCOMPLETE:
```c++
            if (LMIC.dataLen)
             // ....
            else if (LMIC.dataBeg) {
               Serial.print(F("MAC message received: "));
               for (unsigned i = 0; i < LMIC.dataBeg; ++i) {
                   Serial.print(F(" ")); Serial.print(LMIC.frame[i], HEX);
               }
               Serial.println();
            }
            Serial.print("adrAckReq: "); Serial.print(LMIC.adrAckReq); Serial.print("  adrChanged: ");
            Serial.println(LMIC.adrChanged);
```

Now I see:
```console
Packet queued
313384: EV_JOINING
313489: EV_TXSTART
654432: EV_JOINED
netid: 19
devaddr: 260221E6
artKey: ...
nwkKey: ...
654905: EV_TXSTART
826710: EV_TXCOMPLETE (includes waiting for RX windows)
MAC message received:  60 E6 21 2 26 8A 0 0 3 40 2 0 71 3 25 0 FF 1
adrAckReq: -64  adrChanged: 0
2077126: EV_TXSTART
Packet queued
2171404: EV_TXCOMPLETE (includes waiting for RX windows)
MAC message received:  60 E6 21 2 26 8A 1 0 3 40 2 0 71 3 25 0 FF 1
adrAckReq: -64  adrChanged: 0
3421818: EV_TXSTART
```
The MAC messages are:

```
03 40 02 00 71
03 25 00 FF 01
```

These are both LinkADRReqs, decoded as follows.

| DataRate | TxPower | ChMask | ChMaskCtrl | NbTrans |
|:--------:|:-------:|---------|:---------:|:-----:|
|  4 (SF8 / 500 kHz) | 0 (30 dBm) | 0x0002 | 7 (enable 500 kHz, and clear all 125kHz) | 1 |
|  2 (SF8 / 125 kHz) | 5 (20 dBm) | 0xFF00 |  0 (mask for ch 0..15) | 1 |

The spec says "use the global setting of the last command in the group".

So the device should be switching to DR2 (not DR0). But it's not.

Investigating.

terrillmoore, 2019-04-03:
OK, got it. There were two problems: an uninitialized local variable (which explains some of the "why is this happening" problem), and an error that caused `setDrPow()` to always returned "I changed something". My test case formerly was the large MCCI application, which didn't join every time (due to FRAM) so behavior was fundamentally different.

Log of working case:
```console
Starting
Packet queued
313371: EV_JOINING
313476: EV_TXSTART
654545: EV_JOINED
netid: 19
devaddr: 2602263B
artKey: C9-26-8E-59-F-B9-3D-CC-F7-1F-9F-33-F0-6-DD-11
nwkKey: B-F3-EF-40-9D-6-30-E9-41-52-35-C7-A2-C1-28-6B
655087: EV_TXSTART
826912: EV_TXCOMPLETE (includes waiting for RX windows)
MAC message received:  60 3B 26 2 26 8A 0 0 3 40 2 0 71 3 25 0 FF 1
adrAckReq: -64  adrChanged: 1  datarate: 2  adrTxPow: 20
2077337: EV_TXSTART
Packet queued
2149320: EV_TXCOMPLETE (includes waiting for RX windows)
MAC message received:  60 3B 26 2 26 AA 1 0 3 40 2 0 71 3 25 0 FF 1
adrAckReq: -64  adrChanged: 0  datarate: 2  adrTxPow: 20
3399820: EV_TXSTART
Packet queued
3644519: EV_TXCOMPLETE (includes waiting for RX windows)
adrAckReq: -63  adrChanged: 0  datarate: 2  adrTxPow: 20
4894778: EV_TXSTART
Packet queued
5123101: EV_TXCOMPLETE (includes waiting for RX windows)
adrAckReq: -62  adrChanged: 0  datarate: 2  adrTxPow: 20
```
Console log shows the following:

[image]

Examining each packet shows that the uplink rates were SF10, SF8, and finally SF7. This is what we should expect after a join close to a gateway with ADR enabled (which is the default, and has been).  LinkCheckEnable is enabled by default (and this has also been true, although behavior of this has been unpredictable).

terrillmoore, 2019-04-03:
Thanks very much for reporting this. Clearly things are better, but not yet perhaps optimal. More testing is requested.

closed 2019-04-03 by terrillmoore

pomplesiegel, 2019-04-04:
Great, that sort of ADR trajectory is what i'm used to in the near-field as well. I'll be testing this imminently. Thank you!

pomplesiegel, 2019-04-04:
Everything working well with the tip of `master`. Thank you @terrillmoore !

commit a9ccf839 (2019-04-03): Fix #297: uninitialized variable caused unpredictable LADR processing
commit b86590c2 (2019-04-03): Fix #297: don't treat every set of power as a change