mcci-catena/arduino-lmic #298: ADRAckRequest policy is not 1.0.3 compliant

terrillmoore, 2019-04-03:
Hi all,

In reviewing things for responding to #297, I found the following language in section 4.3.1.1, line 458:

> The ADRACKReq SHALL not be set if the device uses its lowest available data rate because in that case no action can be taken to improve the link range.

(This is certainly a great way to learn the spec.)

Of course, the reason we want to send ADRACKReq is to get an indication from the network that we're actually in touch. On TTN, the network sends LinkADRREq, which is perfect; but the spec doesn't require that.

In any case, the spec really doesn't authorize use of `ADRACKReq` for general purposes (after wakeup from power off, and when using the lowest data rate possible. 

The *complliant* thing to do would be to use either `ADRACKReq` or `ACK`, as follows.

- set `ADRACKReq` for link integrity check if DR > DR0. 
- otherwise, if DR == DR0, send message as a confirmed uplink (forcing a downlink).

Based on field experience, we need to have a path that gets us downlink even when at the lowest data rate.

However, confirmed messages have the demerit of having hard scheduling requirements. ADRACKReq is much more network friendly (at least potentially) because the network can be lazy about replying.

I think that our options are:

- Leave it as it is, and send `ADRACKReq` whenever we need confirmation, even at DR0; this works well with TTN (and TTN derivatives), and possibly with other networks.
- Comply with the letter of the spec, but send confirmed messages when we need downlink and haven't heard from the network in "too long"
- Comply with the letter of the spec, and send occasional JOIN requests when we need downlink, are at DR0, and haven't heard from the network in "too long".

My feeling is that (for now) we should leave it as it is.

cstratton, 2019-04-04:
I'd agree with leaving it as is - it seems the perfect example of the not-uncommon case where the LoRaWan spec asks for things that aren't a very good fit for real world needs.

I'd be very wary of unnecessary joins - unless or until there's an actual rejoin message, doing an unnecessary join has the tendency to kill data reporting that _could_ still have worked essentially uplink-only in situations of degraded connectivity, because a server that hears a fresh join request will generate a reply and _abandon any existing session__, regardless if the node hears the join accept and is able to start using the new session, or if it retains only the keys of an old session with which it may keep attempting to send traffic in the hopes that it is being received, even if downlink replies are not.

The beauty that LoRaWan offers in tradeoff for the high cost of downlinks is that done right, it should be possible to obtain at least some uplink capability in a situation where hardly any downlinks get through; ideally even in a situation where _none_ do.

And of course the other problem with using joins for confirmation would be that it leaves the ABP-with-ADR case incompletely working.

terrillmoore, 2019-04-08:
@cstratton I tend to agree; this behavior seems to fit a need.

terrillmoore, 2019-09-01:
Unluckily, the US compliance check (and only that one, as far as I've found) tests exactly this condition. So we have to change policy.  Since we really do want to tell the network that we're going to re-join if the network doesn't get us some traffic, I think the thing to do is to squelch this at the point required by the spec, then turn it on again when we get close to triggering a re-join.

commit 7b7bfdfd (2019-09-04): Fix #298: Fix #411: don't set ADRAckReq at DR0

Clean up documentation, modularize a little, and also reduce the
limit at which we'll rejoin to 720 messages after we hit DR0.
closed 2019-09-04 by terrillmoore