mcci-catena/arduino-lmic #301: Sx126x support has any hope?

addaedans, 2019-04-07:
Hi! Is it possible that there will be support for the SX126x(especially 1262) chip? There is example for ping-pong comm between these chips but it would be quite difficult to implement lorawan into it for me to say the least...The example codes are quite hidden but i can share the links if someone would take a look at it.

terrillmoore, 2019-04-08:
I'm happy to look at the reference code, but I don't have test hardware, or know of a suitable source yet....

terrillmoore, 2019-04-10:
@cstratton found the datasheet for me, thanks. It's not a big project to add support. Steps are:

1. make an abstract API at the radio.c boundary (the radio() function) -- this is just like the multi-region refactoring, use a macro to point to the real driver.
2. edit the current sx1272/6 driver to use a new name for the entry point
3. define HAL ways to get the radio configuration "as wired" pushed into the radio driver.
4. write the driver.

I don't think any changes are needed in lmic*.c for this. It's all radio.c -> radio_sx127x.c and radio_sx126x.c.  I imagine some HAL changes are needed; possibly the whole pinmapping structure has to change because it's 127x-ish.

Next step is to figure out how to find the time (and the suitable reference platform)...

addaedans, 2019-04-10:
Hi! Sorry for the late reply! Here is the link:
http://www.dorji.com/products-detail.php?ProId=65
At the resources tab there is the reference codes link which has all the files for the ping-pong comm. 
Hope you can get it working :)

addaedans, 2019-05-08:
Hi! Is there any progress in the support of the 126x chipset?

terrillmoore, 2019-05-08:
Nope, I don't have hardware, and I don't have budget... So unless we can get someone to offer MCCI support to do this, or someone else wants to do the work, it won't happen for a while. Sorry.

Best regards,
--Terry

addaedans, 2019-05-08:
Hm, i think maybe we can sort this out. If i buy you two 1262 chip and send
to your address it could work?

Terry Moore <notifications@github.com> ezt írta (időpont: 2019. máj. 8.,
Sze 20:09):

terrillmoore, 2019-05-08:
Chips aren't sufficient, alas, we need working systems (CPU + chip + RF, etc.). There's also the issue of funding the engineering work and ongoing support. The cost is likely to be out of reach for an individual, but reasonable for a company. It could be crowd-sourced, I suppose, but I don't have the time to administer that kind of thing; and we spent my pro-bono budget for the year already on [pi64.win](https://pi64.win).

Drop me an email at [tmm@mcci.com](mailto:tmm@mcci.com) and we can discuss further if you're interested.

slavendam, 2021-12-02:
Hi, 
any news about this? 
As I can see, SX1262 became more available and suppliers sending notifications that SX1272 will become outdated soon (not sure what that "soon" really means).

terrillmoore, 2021-12-02:
@slavendam: Technically very possible. No funding. No notifications from MCCI's suppliers about the SX1276, which is what we use. 

As said above:

> There's also the issue of funding the engineering work and ongoing support. The cost is likely to be out of reach for an individual, but reasonable for a company. It could be crowd-sourced, I suppose, but I don't have the time to administer that kind of thing.

I can easily use the Heltec modules for reference. But without a business case to do this, my time is not available. I had no response to the quoted comment previously, so I assume that there is no commercial interest in this. MCCI can't fund it, and I can't fund this myself; the pandemic era has been challenging. Sorry to disappoint. If someone wants to volunteer to run a fundraising drive (via The Things Network NY, which is a non-profit), then it could be crowd sourced. But I don't have the time to run a crowdsourcing program.