mcci-catena/arduino-lmic #309: Continuous integration targets swapped in ci/platformio.sh

bertrik, 2019-04-27:
It looks like the logic to select a target for the continuous integration script is swapped around.

For example, take a look at https://github.com/mcci-catena/arduino-lmic/blob/master/ci/platformio.sh#L18 it checks for the 'avr' TARGET, but then continues to build for the heltec_wifi_lora_32 board, which is an ESP32 board.

Something similar happens at https://github.com/mcci-catena/arduino-lmic/blob/master/ci/platformio.sh#L100 it checks for the 'esp32' TARGET, but builds for the feather32u4 , which is an AVR board.

terrillmoore, 2019-05-17:
Sorry for the slow reply, I was out sick for a few weeks. Will take a look.

terrillmoore, 2019-06-23:
Hm, you're right. I think it's currently just *confusing* rather than *broken* -- the platformio install and tools are totally independent of Arduino, and so it just runs the avr tests during the esp32 phase, and vice versa. But it's awfully confusing and should be fixed.

commit 02c82ef4 (2019-07-04): Fix #309: unswap ci targets in platformio.sh
closed 2019-07-04 by terrillmoore