mcci-catena/arduino-lmic #313: Missing reception of JoinAccept

dberenguer, 2019-05-17:
MCU: STM32L432

Master branch makes my devices miss first three or four JoinAccept packets transmitted by the server. Version 2.3.2 installed from the Arduino IDE does not reproduce this error.

terrillmoore, 2019-05-17:
Which BSP are you using, and which region?  Can you please attach the relevant config header file / -D settings (as you're probably not using a BSP that supports the LMIC configuration directly).

Thanks,
--Terry

dberenguer, 2019-05-17:
Region: CFG_eu868
BSP: Arduino port for the STM32L4 familly: https://github.com/GrumpyOldPizza/arduino-STM32L4

Everything works fine with 2.3.2

dberenguer, 2019-05-17:
My mistake. I was able to reproduce the same behavior on 2.3.2 so it's probably something related to excessive proximity to the gateway, signal saturation or similar.

closed 2019-05-17 by dberenguer