mcci-catena/arduino-lmic #316: Tips for compiling with ESP32 boards?

hsuomal, 2019-05-20:
I'm trying to port code from matthijskooijman/arduino-lmic to this library on TTGO ESP32 board.

I think I changed everything needed in project folder  (eu868 region, 1276 chipset), pin maps from old cold and they should be similar and spi_begin corrected to esp32-style like in old code. From samples I found difference only in network and an event not handled by the older library.

From debug I think it seems to be sending packet, but I get nothing on gateway. Anyone with any tips for this platform? No errors or major symptoms make it hard to figure out where to look next.

Sample debug 1 output:
_21540940: engineUpdate, opmode=0x808
21540965: EV_TXSTART
21541038: TXMODE, freq=868300000, len=23, SF=7, BW=125, CR=4/5, IH=0
Packet queued
start single rx: now-rxtime: 4
21604402: RXMODE_SINGLE, freq=868300000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 21610946 rxtime: 21604394 entry-rxtime: 6552 now-entry: 4 rxtime-txend: 59492
start single rx: now-rxtime: 4
21664150: RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
rxtimeout: entry: 21677734 rxtime: 21664142 entry-rxtime: 13592 now-entry: 4 rxtime-txend: 119240
21885359: EV_TXCOMPLETE (includes waiting for RX windows)
21885362: engineUpdate, opmode=0x800_

terrillmoore, 2019-05-20:
It looks like it's almost working.

1) What radio module are they using? Some modules require activating an antenna switch. HopeRF does not.
2) Check timing. (f you can, pulse a GPIO at txend, and when it starts the receive, and verify that it's a little less than 5 and 6 seconds after tx end).
3) try adjusting the clock tolerance to be sloppier.
4) if you have two devices, use the raw sketch to verify that the radio is working
5) restart your gateway; make sure that it's providing downlink (if you have a known-good device). I've seen a number of problems with that recently.

hsuomal, 2019-05-20:
The same hardware works when compiled with older library. Clock tolerance at +5% already. No antenna switch, sx1276 chips on the board.
Gateway(s) works with this device with older library, compiling with older library traffic is restored. This definately has something to do with library configuration/code. Actually the only difference in  code is the library.
I'm running APB mode ie. no OTAA issues here either. Running Qos0 and no acks/receiving issues should affect this either.

terrillmoore, 2019-05-20:
Well, we know that this library works with *other* ESP32/radio combos. So it's not the library per se.

Are you using the release version (V2.3.2)? (That's what I'd recommend right now.)

If you can post your changes, and a data sheet, I'll be happy to take a look.

hsuomal, 2019-05-20:
I'm using version=2.3.2. That is good to know it has been known to work with other devices. I'll make a simpler sketch to try it out more. Unfortunately TTGO board datasheets are rather poor with no reliable schematics and lots of errors. I'll pick one of the examples for that.
I was initially asking because I'm afraid there is something else than spi initialization that needs to be changed. As is, spi will not start on esp32. I'll be back later on with this.

hsuomal, 2019-05-21:
It seems to be an arduino ide screwup after all. After getting into a minimalistic hello world sample I started getting strange errors and after a full restart and recompile everything started working a treat. Seemed like it was getting parts of old library somewhere.

closed 2019-05-21 by hsuomal

terrillmoore, 2019-05-21:
Bummer when that happens.... thanks for letting me know, and glad it's working for you.