mcci-catena/arduino-lmic #323: LMIC_CONFIG_PRECONDITIONS_H: problem resolving path to lmic_config.h

cyberman54, 2019-05-29:
@terrillmoore

I run in a problem with the mechanics in `LMIC_CONFIG_PRECONDITIONS_H` which fetches  `lmic_config.h` from a build variable. 

In my build script i put the path of `lmic_config.h` in a build variable:

`-D ARDUINO_LMIC_PROJECT_CONFIG_H = '"C:\\Users\\MyUser\\ESP32-Paxcounter\\src\\lmic_config.h"'`

The preprocessor pushes this through

```
#define CFG_TEXT_1(x)	CFG_TEXT_2(x)
#define CFG_TEXT_2(x)	#x
# include CFG_TEXT_1(ARDUINO_LMIC_PROJECT_CONFIG_H)
```

and this resolves to

`c:\Users\MyUser\1-Paxcounter\src\lmic_config.h`

That means the part "ESP32-" is swapped to "1-" in the path for some strange reason, resulting in an invalid path. I assume this is some sort of character escaping problem, but cannot solve it.

cyberman54, 2019-05-29:
addendum: this happenes on a Windows10 system, and it seems it is a case sensitive problem. If i use only lowercase in path, the problem does not occur.

terrillmoore, 2019-05-29:
sorry for slow response. The issue is having the path in the name. The string is getting macro-expanded. `ESP32` is seen as a word, and becomes 1.

The intention was to -D to a name, **without quotes**, and without an explicit path. Use -I to set the include path if necesary. So you'd name the file something unique -- esp32_lmic_config.h, for example, and then say:

`-D ARDUINO_LMIC_PROJECT_CONFIG_H=esp32_lmic_config.h`
`-I "c:/Users/MyUser/ESP32-Paxcounter/src"`

Hope this helps...

terrillmoore, 2019-05-29:
Excellent!