mcci-catena/arduino-lmic #344: Assert error on FSK TX

jpmeijers, 2019-07-13:
As has been stated in the readme FSK has not been tested well and has a few bugs. Therefore this issue should be seen as a note for future reference.

Library version: commit 76faca3

Steps to reproduce:
* Enable an FSK channel after successfully joining the network: `LMIC_setupChannel(8, 868800000, DR_RANGE_MAP(DR_FSK,  DR_FSK),  BAND_MILLI);`
* Force the datarate to be used to FSK: `LMIC_setDrTxpow(DR_FSK,14);`
* Transmit a packet.

```
FAILURE 
/<redacted>/Arduino/libraries/arduino-lmic/src/lmic/radio.c:665
```

I have added a debug print to the code to see what is happening:
```
    // select FSK modem (from sleep mode)
    writeOpmode(0x10); // FSK, BT=0.5
    LMIC_DEBUG_PRINTF("%#02x\n", readReg(RegOpMode));
    ASSERT(readReg(RegOpMode) == 0x10);
```

After setting the OpMode to 0x10, it is read as 0x00. This is why the assert fails.

terrillmoore, 2019-07-13:
hm. ` writeOpmode(0x10); // FSK, BT=0.5` is bogus. That's a reserved value on the 1276.  This is a holdover from the 1272, where it selects (per the comment) gaussian filter BT=0.5.

The correct value to write is 0x00 (on the 1276). I suspect that the FSK needs a careful path reviewing for 1272-isms, as there were a number of them in the LoRa code that I caught in previous bugs.

jpmeijers, 2019-07-14:
I changed the FSK TX code to look like this:
```
    // select FSK modem (from sleep mode)
    //writeOpmode(0x10); // FSK, BT=0.5
    //ASSERT(readReg(RegOpMode) == 0x10);
    opmodeFSK();
    ASSERT((readReg(RegOpMode) & OPMODE_LORA) == 0);
```

Using an RTL-SDR I can see the packets are being transmitted. Looking at my gateway's UDP traffic I can also see that it received the packets and forwarded them to TTN. TTN however does not seem to accept the FSK packets.

```
jpmeijers@jpmeijers-XPS13-9333:~$ netcat -ul 1700
g��'���؃u{"time":"2019-07-14T14:24:27Z","rxpk":[{"tmst":6945716,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-53,"size":26,"data":"QDskASbAWAABufl0LGVaV5rVMOz3997J1xc="}]}is�'���؃u{"time":"2019-07-14T14:24:34Z","rxpk":[{"tmst":14093845,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-57,"size":26,"data":"QDskASbAWQABk4m1f31lA0RlM5uulHqp5a0="}]}J��'���؃u{"time":"2019-07-14T14:24:41Z","rxpk":[{"tmst":21241966,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-55,"size":26,"data":"QDskASbAWgABkJ4WTGkTpgXzuIVsdcC0XKc="}]})��'���؃u{"time":"2019-07-14T14:24:48Z","rxpk":[{"tmst":28392426,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-57,"size":26,"data":"QDskASbAWwAB+j444xTe9pLk9Cwlplk758Q="}]}���'���؃u{"time":"2019-07-14T14:24:52Z","stat":{"time":"2019-07-14 14:24:52 GMT","lati":-33.93597,"long":18.87081,"alti":0,"rxnb":5,"rxok":4,"rxfw":4,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"IMST + Rpi","mail":"","desc":"JPM <Pi Supply shield>"}}�F�'���؃u{"time":"2019-07-14T14:24:55Z","rxpk":[{"tmst":35540557,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-59,"size":26,"data":"QDskASbAXAABnIbE6m45vyvu4U/o2lXvdko="}]}|��'���؃u{"time":"2019-07-14T14:25:02Z","rxpk":[{"tmst":42688689,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-57,"size":26,"data":"QDskASbAXQABg2D1BnuNCj/APLxc1WSg6U0="}]}T��'���؃u{"time":"2019-07-14T14:25:05Z","rxpk":[{"tmst":45424291,"chan":4,"rfch":0,"freq":867.300000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.8,"rssi":-72,"size":23,"data":"QIUYASYAnQACV/orE3Q3sUPy7bOftas="}]}��'���؃u{"time":"2019-07-14T14:25:09Z","rxpk":[{"tmst":49836819,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-55,"size":26,"data":"QDskASbAXgABR8uVhm5eDDvkFnbRLDN3YYo="}]}���'���؃u{"time":"2019-07-14T14:25:17Z","rxpk":[{"tmst":56984945,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-55,"size":26,"data":"QDskASbAXwABfse03WjTfJ4tBSZaNGTSTms="}]}.c�'���؃u{"time":"2019-07-14T14:25:22Z","stat":{"time":"2019-07-14 14:25:22 GMT","lati":-33.93597,"long":18.87081,"alti":0,"rxnb":7,"rxok":5,"rxfw":5,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"IMST + Rpi","mail":"","desc":"JPM <Pi Supply shield>"}}3��'���؃u{"time":"2019-07-14T14:25:24Z","rxpk":[{"tmst":64133063,"chan":9,"rfch":1,"freq":868.800000,"stat":1,"modu":"FSK","datr":50000,"rssi":-54,"size":26,"data":"QDskASbAYAABOGqGMBsuWFqkybMUODvaCmk="}]}
```

Also see:
https://github.com/TheThingsNetwork/ttn/issues/766

terrillmoore, 2019-08-17:
Turns out that the RWC5020A will test FSK as part of certification. So I'm about to fix this and #346.

terrillmoore, 2019-08-17:
(pushed wrong button, didn't mean to close)

commit 5a37419b (2019-08-21): Fix #344: Fix #346: improve FSK handling, add docs
closed 2019-08-24 by terrillmoore