mcci-catena/arduino-lmic #350: Help With Example: ttn-otaa-network-time

omaranw, 2019-07-16:
Hi,

In the 'user_request_network_time_callback' function of example https://github.com/mcci-catena/arduino-lmic/blob/master/examples/ttn-otaa-network-time/ttn-otaa-network-time.ino,

I am unable to get a non zero value of flagSuccess (line 214 and 221). Any idea why that can be? I have just started working on LoRaWAN and I am trying to fetch time so I can timestamp my data (without using RTC module). The communication with TTN is working without any problem. I am using dragino LoRa Shield and Sparkfun Redboard Turbo.

Will appreciate a pointer or two on what I should try to get this example working.

Thanks

omaranw, 2019-07-16:
[image]

Here is a snip of relevant LMIC documentation just in case.

frazar, 2019-07-16:
Is it possible that you're missing the following definition in project_settings/lmic_project_config.h ?

```C
#define LMIC_ENABLE_DeviceTimeReq 1```

Unfortunately, his is mentioned in the `README` but not in the .doc document.

omaranw, 2019-07-16:
Well pointed as I was missing it, and I just tried this definition as well. The only change I have seen is that the error message appears after some delay now. So it is trying to fetch the time, but is still not successful.

frazar, 2019-07-16:
Is your gateway equipped with a GPS module?

If not than I suggest you follow the workaround in the section named "Network requirements"
 of the description to PR #147.

terrillmoore, 2019-07-16:
> The communication with TTN is working without any problem.

Hm, TTN doesn't support network time request, as far as I know. That may be your other problem.

omaranw, 2019-07-16:
> Is your gateway equipped with a GPS module?

I am not sure, but my guess is it isn't. My university has a few gateways in the surroundings, and some of them not owned by university. I will have a read of workaround you shared and will get back to you.

Thanks for the help

omaranw, 2019-07-16:
> If not than I suggest you follow the workaround in the section named "Network requirements" of the description to PR #147.

Just to make sure I understood it properly, the alternate is to use the custom packet forwarder on gateway. As I don't own the gateway, I believe this is a dead end for me. I originally started by trying to fetch timestamp at TTN console and somehow send it back as data to the node. Someone pointed out this alternative, sadly gateways are not something I can control. Lets see if I can find another way.

omaranw, 2019-07-16:
> Hm, TTN doesn't support network time request, as far as I know. That may be your other problem.

I started this journey with the assumption that fetching time cannot be hard. Guess I was wrong, very wrong.

frazar, 2019-07-16:
> Just to make sure I understood it properly, the alternate is to use the custom packet forwarder on gateway.

Yes.

Unfortunately, you need a cooperating gateway to get accurate time syncronization.

terrillmoore, 2019-07-31:
@omaranw please check this comment: https://github.com/mcci-catena/arduino-lmic/issues/268#issuecomment-474470323

@cyberman54 has an application-level approach to getting time to the device; it works with TTN v2. You should check that.

omaranw, 2019-08-01:
@terrillmoore Thanks for referring to that comment. I have asked the author to explain if possible. As far as I understand, it is the local time of device, not the network time.

cyberman54, 2019-08-01:
You can't derive time-of-day from LMIC.txend. You need a loop over an application server which sends timestamp, then calibrate looptime by using LMIC.txend.

Study my example in paxcounter timesync.cpp how to do it.

Of course it's easier to use LoRaWANn time request, but you need a network which supports this. TTN v2 does not.

omaranw, 2019-08-01:
@cyberman54 thanks, I will have a look.

terrillmoore, 2019-08-01:
Thanks @cyberman54 !

terrillmoore, 2020-03-04:
@cyberman54 Awesome, thanks!  @frazar Great work! So ... is it time to close this?

cyberman54, 2020-03-04:
Yes, we can close this, from my opinion. The example is working and complete.

terrillmoore, 2020-03-04:
OK, closing, thanks everyone for the work on this!

closed 2020-03-04 by terrillmoore